fritzing/fritzing-app #3920: Pin label offset on generic IC

DarkLiKally, 2022-01-02:
Hello,
I wanted to pick up the issue fritzing/fritzing-parts#227 again because I've more information about the error posted there after it was closed and the issue is still a problem for other users. There was sadly no reaction there, so I decided to create a new issue.

Please read the above mentioned issue first. Here are my additional informations:

I've the same issue with the latest version of fritzing.
[image]

Steps how it appeared for me: 
1. Create generic IC
2. Adjust the pin labels (Edit pin Labels)
3. designed some circuit
4. went into PCB view and moved stuff around and routet some lines
5. deleted the IC by accident (wanted to delete a routing)
6. pressed strg + z
7. The labels went offset, also labels were reset 
8. Set labels again
9. I've the result seen in the screenshot above

UPDATE:

I've fixed the problem for me by opening the project file with an archive editor and extracting the xml file (fz-file).

I've edited the geometry part of the part for schematic view. I have a second copy of the IC on my project so I compared the two, because the second one was still working. I've noticed that the attributes m31 and m32 were different for the two parts. I've copied over the values from the working one, saved and repacked the archive. The component was slightly moved to the side when reopening the projct in fritzing, but no problem, the Labels were placed correctly now. I've dragged the component back into it's desired place and everything looks fine now.

                    <geometry z="2.50364" x="157.5" y="103.5">
                        <transform m11="0" m12="1" m13="0" m21="-1" m22="0" m23="0" m31="112.5" m32="-31.5" m33="1"/>
                    </geometry>

Hope this information helps you to find and resolve the issue.

KjellMorgenstern, 2022-01-11:
Closing as duplicate of #3180

closed 2022-01-11 by KjellMorgenstern

failiz, 2022-01-11:
This bug, in addition to the offset, also mentions that the labels are reset to their default state after a CTR+Z command. This is a bug that still needs to be fixed.
Steps:
Drag an generic IC 
Edits its labels
Delete it
Undo the last command.

The labels of the IC are the default ones, not the edited ones.

KjellMorgenstern, 2022-01-11:
@failiz Yes, the issue could (and probably should) have been split into several others. State was also influenced by layer visibility settings in the "view" menu. Should all be covered.