fritzing/fritzing-app #3922: Swapping stepper motors messes up connections

KjellMorgenstern, 2022-01-12:
## Current Behaviour

When swapping bipolar and unipolar steppers, connections are messed up:

Bipolar:
[image]

Then swap to unipolar:
[image]

close up:
[image]

## Expected Behaviour
The connection should be separated. Also possible: The motors are not interchangeable and can not be swapped.
Possible future solution: Adding cable and connector features to Fritzing would also solve this.

failiz, 2022-01-12:
"The motors are not interchangeable and can not be swapped." The Arduino family can be swapped and they are not interchangeable. If you add an Arduino nano to the breadboard and then swap it to an Arduino Uno, cables appear.

failiz, 2022-01-19:
Maybe the solution is to split the family in two: bipolar stepper motors and unipolar stepper motors. This is what is has been done for the switches: 
* switch (for four pin switches)
* switch - 2 pin
* switch - 3 pin

In any of the these two categories (bipolar and unipolar) we could have more options: Nema-8, Nema-11, Nema-14, Nema-17, Nema-23, and other types of permanent magnet stepper motors.

mMerlin, 2022-01-19:
4 Wire and 6 wire servos are obviously not "interchangeable". That does not mean they need to be in a separate part family. Swapping parts via Inspector is (also) a convenience when initially *choosing* a part. That is before any wires are connected to that. For that scenario, swapping **similar** (not directly interchangeable) parts works just fine. Same for the different Arduino boards that have different physical sizes and layouts.

It is up to the user to make adjustments for those cases. If you change the design of a project part way through, you should **expect** to have some rework to do.

Now, if you want to propose an extension for inspector, maybe a checkbox could be added to filter the swapable parts. To only offer parts with the same number of connections. Or even smarter, to only offer parts with pins that match the existing wires. But do not force family groups to be broken up to "allow" one to one swapping. Every Arduino board would have be its own family, and they would all have to be separately visible in the parts bins.

failiz, 2022-01-21:
I guess it is a trade-off, more families simplify the code (especially for dynamically generated parts like headers) and they create fewer complications when swapping them. On the other hand, a lot of families make it difficult to find other similar parts (and change to them) and complicate the part management. There were no rules about that and there are several parts which could be merged in one family: e.g. switches (2, 3 and 4 pin), transistors (npn, pnp, mosfet), inductors (air-core inductor, sparkfun inductor), etc.). This is especially true for Sparkfun parts which have different properties ModuleIDs and families.
In any case, I am not suggesting that we should split all the Arduinos in different families. Just that we should define some families which should have standard properties and that we should try to fix the parts that we have in the core to comply with these rules.

In any case, another solution for the stepper motors is could be that the connections are modified to be straight. So, unipolars motors will have A+, Amid, A-, B+, Bmid, B- and bipolar motors will have A+, empty, A-, B+, empty, B-. That will avoid issues when swapping the parts. When going from bipolar to unipolar the swapping will add the new connectors and they will not overlap with the previous ones and when swapping from unipolar  to bipolar some connections will be removed.

KjellMorgenstern, 2022-01-21:
I think modifying the connectors goes into the right direction.
We are thinking about adding a "Connector" class, which would only connect to other connectors with the same type property.
For example, JST-XH-4 would not connect to JST-XH-6 (or 3). 
So, when swapping such parts, the connections would just be deleted. Of course this would be undoable with Ctrl-Z

failiz, 2022-01-21:
Class or property? If it is a class, I do not get what you are planning. If it is a new property, I would avoid to call it JST-XH-4 as there are several connectors that you can use for the motors. And a lot of motors come without a connector (there are wires coming out of the connector). 

I think that the motors should be as generic as possible. Maybe the property could be number of connectors and the value will be 4 for bipolar motors and 5 or 6 for unipolar. 

In any case, I do not think that this property will fix the problem. The problem comes because the position of the connectors overlap in the breadboard view.