fritzing/fritzing-app #3927: Specify a connector as an origin for each view in the part

failiz, 2022-01-19:
## Problem
The current part format does not specify an origin for the part. Fritzing uses the upper left corner of the svg for positioning the part. If you drag a part from the bin and the align to grid is enabled, Fritzing aligns one of the connectors to the grid. However, when there is a swap, the position of the old part is maintained. This does not make sense as all the pins can be not aligned. 
See the example with a potentiometer:
Before is swapped all the pins are aligned to the grid:
[image]

After swapping to a rotary potentiometer, all the pins are not aligned:
[image]

If we place another trimmer potentiometer overlapping the rotary potentiometer, we can see how the upper left corner is at the same place:
[image]

Issue #3077 suggests to align to the grid after swapping and rotating, but this is not enough for swapped parts. If the new part is offset 0.05in, the aling to grid could decide to align to two different positions in the grid. In addition, when obsoleting parts, it could be that the size of the parts has changed. Having a connector that specifies what the origin is would allow having at least one connector placed at the correct position.

## Proposed Solution
Save in the part file a connector that will be used as an origin for the part. This was suggested by @mMerlin in #299.
> The uno and other boards that use the same header spacing have the same problem. Even more so, when the ICSP headers are included. What I think is really needed, is a way to specify which connector in the part to use as the 'origin', to align to the grid. That way the part creator can choose which block(s) of pins align, and which don't. Like:
> 
> ```
> <connector id="connector9" origin="true>
> ```
> 
> or
> 
> ```
> <connector id="connector9">
>   <breadboardView>
>     <p layer="breadboard" origin="true" …/>
> ```
> 
> These cases are all microcontroller boards, and are not normally placed on breadboards. They typically (should) have type="female" connectors, so that parts (shields) can connect to them. The alignment is still annoying, because wires do not connect "straight" to (some of) the pins. No really "good" fix, since want to keep the part (bb view) matching the physical board. It would be possible to create variant parts where the bb view connectors are deliberately not where they are on the physical board. That would look a bit cleaner when running wires directly to the pins. It would also break if attempting to position an "accurately" designed shield/hat board over top.
> 
> Discontinued parts may not get 'priority', but they can still be useful. They still exist 'in the wild'.

KjellMorgenstern, 2022-01-19:
Fritzing is already using the connectors to align parts.
However, right now, Fritzing always chooses the first connector.
As I understood, you suggest a feature, where this connector can be manually specified by adding the origin attribute, right?

failiz, 2022-01-19:
Yes, but not only that. I think that that pin marked as origin should be kept in the same position when swapping parts. 
If we have an obsolete part where the size of the svg has changed or the position of the connectors have changed, the part will not be aligned correctly, see above example. The connectors of the rotary pot are far from the connectors of the trimmer pot. I think that at least one should match. The aliment to the grid could help, but it could not be enough in places where the svgs are very different. 
Think in two parts with 3 pins spaced 0.1inches but with difference svg sizes where the connectors are placed in different places. Specifying a connector as the origin, would make the swapping much easier. If these parts where connected to a breadboard, the swapping mechanism would add wires to keep the connections. If the origin is preserved, no wires would be added. 

Maybe there are two separate issues here. One is to specify a connector to be the origin as you suggest, which will help to specify which is the pin aligned to the grid (in cases where there connector distances are not equal or a multiple of the grid size). And the other is to use the origin (or first connector if not specified) of the parts for swapping. In that case, the position of the origin connector would not change in the sketch, making swapping parts less problematic. See the discussion of problems that a small change of size in a svg causes when obsoleting a part in https://forum.fritzing.org/t/potentiometer-symbol/14790/38

KjellMorgenstern, 2022-01-19:
Isn't the alignment after swapping the same as #3922 ?
Then this issue should only be about specifying origins different from the first connector.

mMerlin, 2022-01-19:
"Fritzing is already using the connectors to align parts."
That does not appear to be true for the specific context of swapping parts from the same family using Inspector. As shown in the images in the original post. The upper left corner of the old and new svg of the 2 parts are "aligned". Not a connector.

Swapping parts that already have wires connected have additional constraints, and problems, if the number of connectors is different. Or different order or relative positions.