fritzing/fritzing-app #3931: A part name with a forward slash will crash Fritzing on macOS

yamadapc, 2022-02-12:
## Current Behaviour
A part name with a forward slash will crash Fritzing on macOS

## Expected Behaviour
The app will not crash.

KjellMorgenstern, 2022-02-12:
Just works as expected here when adding a forward slash to a part title

Which version of Fritzing are you using?
Did you mean the title (Fritzing parts don't have a 'name' field) ?

yamadapc, 2022-02-12:
I’m not sure. I opened part edit, clicked save as new part. Typed “1/4 …” and clicked save.

KjellMorgenstern, 2022-02-12:
Can you sent a screenshot of the version? You can see it in the "About" dialog.

yamadapc, 2022-02-12:
Version 0.9.9

It's relatively clear this wouldn't work, but:

[image]

```
Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000050
Exception Codes:       0x0000000000000001, 0x0000000000000050
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [82611]

VM Region Info: 0x50 is not in any region.  Bytes before following region: 4390707120
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      105b4e000-10641e000    [ 9024K] r-x/rwx SM=COW  ...acOS/Fritzing

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   Fritzing                      	       0x105ed08f4 ModelPart::setAlien(bool) + 4
1   Fritzing                      	       0x105c4a2bd PEMainWindow::saveAs(bool) + 10061
2   Fritzing                      	       0x105f95fe1 0x105b4e000 + 4489185
3   QtCore                        	       0x10882a075 0x1085fe000 + 2277493
4   QtWidgets                     	       0x106de4406 QAction::activate(QAction::ActionEvent) + 310
5   QtCore                        	       0x108821b3f QObject::event(QEvent*) + 943
6   QtWidgets                     	       0x106deda2a QApplicationPrivate::notify_helper(QObject*, QEvent*) + 266
7   QtWidgets                     	       0x106deee51 QApplication::notify(QObject*, QEvent*) + 497
8   Fritzing                      	       0x105b831aa FApplication::notify(QObject*, QEvent*) + 26
9   QtCore                        	       0x1087f69d4 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 212
10  QtCore                        	       0x1087f7d19 QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 809
11  libqcocoa.dylib               	       0x1073d4189 0x10739a000 + 237961
12  libqcocoa.dylib               	       0x1073d48f8 0x10739a000 + 239864
13  CoreFoundation                	    0x7ff81bc088fd __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
14  CoreFoundation                	    0x7ff81bc08865 __CFRunLoopDoSource0 + 180
15  CoreFoundation                	    0x7ff81bc085e4 __CFRunLoopDoSources0 + 242
16  CoreFoundation                	    0x7ff81bc0701b __CFRunLoopRun + 893
17  CoreFoundation                	    0x7ff81bc065dd CFRunLoopRunSpecific + 563
18  HIToolbox                     	    0x7ff8248434f1 RunCurrentEventLoopInMode + 292
19  HIToolbox                     	    0x7ff824843118 ReceiveNextEventCommon + 284
20  HIToolbox                     	    0x7ff824842fe5 _BlockUntilNextEventMatchingListInModeWithFilter + 70
21  AppKit                        	    0x7ff81e635d88 _DPSNextEvent + 886
22  AppKit                        	    0x7ff81e6343f4 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1411
23  AppKit                        	    0x7ff81e626919 -[NSApplication run] + 586
24  libqcocoa.dylib               	       0x1073d355f 0x10739a000 + 234847
25  QtCore                        	       0x1087f2a6f QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 431
26  QtCore                        	       0x1087f6fe2 QCoreApplication::exec() + 130
27  Fritzing                      	       0x105b937ca main + 298
28  dyld                          	       0x10d7054fe start + 462
```

KjellMorgenstern, 2022-02-12:
Ah, got it. The prefix will be part of the filename, and forward slashes can not be used for that.
Anyway, Fritzing could sanitize the input to avoid this.