fritzing/fritzing-app #3932: Copper fill is not complet in combination with curved traces

ClemensGruber, 2022-02-17:
I would expect a filled copper section near to the traces as done in some cases. But a lot of space is not filled out with copper: 

[image]

failiz, 2022-02-17:
It seems that the ratsnets are blocking the copper fill. Could you also attach the file? If not, could you check if there are ratsnets around where the copper stops?

KjellMorgenstern, 2022-02-17:
[image]

I think Bezier curves are not handled correctly.

ClemensGruber, 2022-02-22:
@failiz seems _not_ to be a problem of a ratsnet or other "special" traces but a general problem with Bezier traces as @KjellMorgenstern pointed out. Attached a realy most simple demo - `*.fzz ` files are not allowed here, so I have packed it in a zip - but you can easily reproduce it by yourself in Fritzing: Make a trace, press CTRL and drag'n'drop the trace to make it curvy. 

I wonder why it has not shown up in the past, it's so obvious! Had it worked with older versions? I'm using the latest Fritzing 0.9.9 

[copper-fill-demo-bezier-traces.zip](https://github.com/fritzing/fritzing-app/files/8116303/copper-fill-demo-bezier-traces.zip)

[image]

cgruber-uos, 2023-02-12:
Do you have any hint to overcome this by a workaround - adding additional (usless) )traces, set ground points ... so my only solution would be have no ground fill at all so that the PCB shows not so "damaged".

KjellMorgenstern, 2023-02-12:
Far from a solution, not even a workaround, just somewhat mitigating the issue. You can split the curve into multiple curves and lines. Here I used 2 lines and 2 curves instead of the curve in the minimal demo. 

[image]

ClemensGruber, 2023-08-13:
Many thanks for adding this to the 1.0.0 milestone and fixing it with the release! Looks much better now! 

[image]