fritzing/fritzing-app #3935: the bottom side Label on two sided PCB is showing as mirror, and there is no option to flip it over, the problem has seen into Gerber file.

TonyCham6, 2022-03-01:
## Current Behaviour
Text
<please attach screenshot if possible>
**Build:**
<e.g. Version 0.9.3 (b5c895d32 2016-04-19) Cocoa [Qt 5.12.2]

**Operating System:**
<Windows 10>

**Steps to reproduce:**
- Step 1
- Step 2

## Expected Behaviour
Text
the bottom side Label on two sided PCB is showing as mirror, and there is no option to flip it over, the problem has seen into Gerber file.

[image]
[image]

KjellMorgenstern, 2022-03-01:
Is this a report for a six year old version of Fritzing?

TonyCham6, 2022-03-02:
> Is this a report for a six year old version of Fritzing?

HI ,
Not really, it's very recent downloaded !!!
[image]
Regards.

KjellMorgenstern, 2022-03-02:
Which version is it? I am asking since you labeled the report for 0.9.3b

TonyCham6, 2022-03-03:
Hi,
It is 0.9.9
Thanks

vanepp, 2022-05-29:
I believe this is working as designed (and as expected.) Here I put an SMD resistor R1 on the top of the board and another one R2 on the bottom of the board. In the gerber output the top silkscreen layer has the R1 for the top of the board and the bottom silkscreen layer has the inverted R2 (because we are looking from the top of the board) on the bottom silkscreen layer so it will be printed the correct direction on the bottom of the board. This image is the gerber output displayed with gerbv. 

[image]

I'm assuming that batt+ in the above image is an SMD part and on the bottom of the board (that isn't clear from only the images, the .fzz file is a more useful thing to post!)