fritzing/fritzing-app #3936: Browser version of Fritzing with WebAssembly?

Chris2011, 2022-03-02:
## Problem
Atm I need to install Fritzing as a Desktop application, which is good but I like the idea that a lot of applications can run in the browser, just smoothly. So I thought by myself, is there a web version of this? And if not, is it possible to have one?

## Proposed Solution
For this, webassembly came into my mind and I found this documentation about Qt and WebAssembly: https://doc.qt.io/qt-5/wasm.html
So at the end, I think this could be possible, but I've no idea how much work it will be to clean up the code to be compatible with WebAssembly.

If someone already tried this, please share your infos. When I have time I will also try to play with Qt for WebAssembly and will see what problems I have then.

failiz, 2022-03-10:
As far I know, there is no browser version of Fritzing. If you try this, please, keep us updated!

Chris2011, 2022-03-10:
Will just play around, will be my first impression with WebAssembly :)

KjellMorgenstern, 2022-03-10:
We had a look at WebAssembly several months ago, but it didn't seem fit. Closing this for now.

closed 2022-03-10 by KjellMorgenstern