fritzing/fritzing-app #3938: Crash when switching part 'package' option

yamadapc, 2022-03-08:
## Current Behaviour
Fritzing will crash after switching package a couple of times for a part

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Fritzing [85274]
Path:                  /Applications/Fritzing.app/Contents/MacOS/Fritzing
Identifier:            org.fritzing.Fritzing
Version:               0.9.9 (bda4e)
Code Type:             X86-64 (Translated)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-03-09 07:59:45.9068 +1100
OS Version:            macOS 12.2.1 (21D62)
Report Version:        12
Anonymous UUID:        538D25F5-BE8A-C61A-5156-9B98071AB86A

Sleep/Wake UUID:       9C7B03EE-3F05-4A47-AC0E-E514DF564AAA

Time Awake Since Boot: 140000 seconds
Time Since Wake:       1377 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000020
Exception Codes:       0x0000000000000001, 0x0000000000000020
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [85274]

VM Region Info: 0x20 is not in any region.  Bytes before following region: 4369629152
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      104734000-105004000    [ 9024K] r-x/rwx SM=COW  ...acOS/Fritzing

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   Fritzing                      	       0x104a8e854 ConnectorItem::attachedToViewLayerID() + 4
1   Fritzing                      	       0x104b2005e SketchWidget::makeSwapWire(SketchWidget*, ItemBase*, long, ConnectorItem*, ConnectorItem*, Connector*, QUndoCommand*) + 302
2   Fritzing                      	       0x104b1da73 SketchWidget::setUpSwapReconnect(SwapThing&, ItemBase*, long, bool) + 2659
3   Fritzing                      	       0x104b1cb45 SketchWidget::setUpSwap(SwapThing&, bool) + 133
4   Fritzing                      	       0x1047a2cf1 MainWindow::swapSelectedAuxAux(ItemBase*, QString const&, ViewLayer::ViewLayerPlacement, QMap<QString, QString>&, QUndoCommand*) + 1105
5   Fritzing                      	       0x1047a1bf7 MainWindow::swapSelectedAux(ItemBase*, QString const&, bool, ViewLayer::ViewLayerPlacement, QMap<QString, QString>&) + 567
6   Fritzing                      	       0x1047a146d MainWindow::swapSelectedMap(QString const&, QString const&, QMap<QString, QString>&, ItemBase*) + 4477
7   Fritzing                      	       0x1047a0241 MainWindow::swapSelectedTimeout() + 193
8   QtCore                        	       0x11147d075 0x111251000 + 2277493
9   QtCore                        	       0x111486462 QTimer::timerEvent(QTimerEvent*) + 98
10  QtCore                        	       0x1114747ff QObject::event(QEvent*) + 111
11  QtWidgets                     	       0x10e978a2a QApplicationPrivate::notify_helper(QObject*, QEvent*) + 266
12  QtWidgets                     	       0x10e979e51 QApplication::notify(QObject*, QEvent*) + 497
13  Fritzing                      	       0x1047691aa FApplication::notify(QObject*, QEvent*) + 26
14  QtCore                        	       0x1114499d4 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 212
15  QtCore                        	       0x1114af2ff QTimerInfoList::activateTimers() + 991
16  libqcocoa.dylib               	       0x110c42822 0x110c0a000 + 231458
17  CoreFoundation                	    0x7ff807790c08 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
18  CoreFoundation                	    0x7ff807790b70 __CFRunLoopDoSource0 + 180
19  CoreFoundation                	    0x7ff8077908e3 __CFRunLoopDoSources0 + 242
20  CoreFoundation                	    0x7ff80778f2ff __CFRunLoopRun + 897
21  CoreFoundation                	    0x7ff80778e8a9 CFRunLoopRunSpecific + 567
22  HIToolbox                     	    0x7ff81081a4f1 RunCurrentEventLoopInMode + 292
23  HIToolbox                     	    0x7ff81081a247 ReceiveNextEventCommon + 587
24  HIToolbox                     	    0x7ff810819fe5 _BlockUntilNextEventMatchingListInModeWithFilter + 70
25  AppKit                        	    0x7ff80a0e4d88 _DPSNextEvent + 886
26  AppKit                        	    0x7ff80a0e33f4 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1411
27  AppKit                        	    0x7ff80a0d5919 -[NSApplication run] + 586
28  libqcocoa.dylib               	       0x110c4355f 0x110c0a000 + 234847
29  QtCore                        	       0x111445a6f QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 431
30  QtCore                        	       0x111449fe2 QCoreApplication::exec() + 130
31  Fritzing                      	       0x1047797ca main + 298
32  dyld                          	       0x20548a4fe start + 462
```

**Operating System:**
macOS

**Steps to reproduce:**
- Add rotary encoder to breadboard
- Switch package a couple of times

## Expected Behaviour
No crash

failiz, 2022-03-10:
The rotary encoder from the core bin does not allow to switch to different packages. Could you specify which rotary encoder you are using?

KjellMorgenstern, 2022-04-19:
Probably a duplicate of #3863 
Also, it was likely the Sparkfun encoder

closed 2022-04-19 by KjellMorgenstern