fritzing/fritzing-app #3940: "Net label" text is cut off in Schematic view

fluidblue, 2022-03-20:
## Current Behaviour
A net label's text is cut off, when the text is longer than a few characters.
[Screenshot](https://imgur.com/a/IWb3j4Y)

**Build:**
Version 0.9.9 (bCD-348-0-f0af53a9 2021-09-22) Cocoa [Qt 5.15.2]

**Operating System:**
macOS 11.6.3 Big Sur

**Steps to reproduce:**
- Create a new project and go to Schematic view
- Search parts for "net label" and add a new "net label".
- Enter long text in the label property, e.g. "lorem ipsum dolor sit amet"

## Expected Behaviour
The net label's text should be shown completely.

failiz, 2022-06-19:
I could not reproduce this in 0.9.9 or 0.9.10 (Windows). Is this a MacOS specific bug? 
[image]

KjellMorgenstern, 2022-11-16:
Imported screenshot link from imgur
[image]

KjellMorgenstern, 2023-09-08:
Comparing Macos (above) and Linux (below).
Both have roughly the same font size at 990% zoom, but the NetLabel on MacOS is too small.

[image]

failiz, 2024-01-04:
I wonder if we could change to a monospaced font to make our life easier and fix this problem and https://github.com/fritzing/fritzing-app/issues/3630.

KjellMorgenstern, 2024-01-05:
Fritzing has a method to calculate bounding boxes in SVGText. This is implementation is probably now obsolete: The missing bounding box implementation was added to Qt with v6.2. 

Most of these kind of bugs are not related to the font width directly, but svg transforms, styles, using a fallback font (this one looks suspiciously like a fallback font that is simply not installed on macos) .

KjellMorgenstern, 2025-07-28:
This was fixed in 1.0.5. 
The default font for netlabels is now Noto Sans. Old sketches will still have Open Sans. You can choose the fonts in the inspector in case you are working with old sketch.
Font size calculations have been tuned OS specific for some SVG related rendering issues, and try to use metrics from the OS font engines.