fritzing/fritzing-app #3945: can not open the component library

NOmansky12, 2022-04-22:
## Current Behaviour
Text
<please attach screenshot if possible>
**Build:**
<e.g. Version 0.9.3 (b5c895d32 2016-04-19) Cocoa [Qt 5.12.2]

**Operating System:**
<Windows 10>

**Steps to reproduce:**
- Step 1
- Step 2

## Expected Behaviour
Text

NOmansky12, 2022-04-22:
i try anyother version and i already has complete component library in my computer,but i cant open the component library

vanepp, 2022-04-25:
There isn't enough information here to do anything useful. What Fritzing version and what operating system are you using.? How did you load Fritzing, from the Fritzing.org site or from a OS specific repository? You may also be better off asking about this (with the same information!) in the forums at 

https://forum.fritzing.org/

closed 2022-05-10 by KjellMorgenstern