fritzing/fritzing-app #3948: Allow user color choices to differentiate top/bottom copper traces on PCB

Reynolds-Beta, 2022-05-10:
## Problem
Top and bottom copper traces are very similar in color. Color-blind folks can find these colors to be too close to distinguish, particularly when switching between top and bottom views.

## Proposed Solution
Allow user defined color choice.

chandradharkoneti, 2023-07-18:
I am having the same problem.

chandradharkoneti, 2023-07-18:
This would be a very useful feature to have since I have trouble differentiating close colours. If I had the option to change top layer traces to black and bottom layer traces to light blue and jumpers to red.

Or some other customisable colour scheme that would be friendly to partly colour-blind people. It would be of great help. Currently, the top layer copper is gold colour and bottom layer copper is orange colour. Since those two colours are close to each other, I am having trouble telling whether a trace is on the top layer or bottom layer.

I know that this was done so that the PCB looks realistic whilst designing, but an un-intended consequence of that is the colours are too close to each other.

mapsedge, 2023-09-21:
PCB view: Jumping in on this one. The ability to apply a custom color to any trace would be very helpful in keeping track of where things are going. It's obvious that Fritzing can identify individual segments of traces (e.g. Delete up to bendpoint), so perhaps a new context item "Color" with a color selector flyout. 

A simple, select-based color picker with, say, 16 colors would be enough, or include a full palette with the ability to save colors for frequent use.

Although I can't imagine anyone wanting this particular variation, maybe the flyout could include two choices, "Entire Trace" and "Selected Segment."