fritzing/fritzing-app #3950: Provide more information about why a custom part cannot be loaded.

mikedehaan, 2022-05-19:
## Problem
When creating a new part, the system can fail with a generic message stating "Unable to load part from '<PATH TO FZPZ PARTS FILE>'"

## Proposed Solution
Either print the reason for failure or add a more detailed message to the Debug log window when enabled. File name and line numbers would be ideal.

## Reason
Creating a part can be a complex process and can fail in many ways. A single generic message is often not enough to help the designer fix issues.

vanepp, 2022-05-29:
A fix can be as simple as adding "no part.part-name.fzp file found in the .fzpz file" to the error message, since that was the problem here. The fzp file in the .fzpz did not start with part. and thus there was indeed no part to load. In most other cases I am aware of it will provide a file name where the error occurred. Here it didn't have a file name to provide.

KjellMorgenstern, 2023-08-23:
More detailed information is given now, for example:
[image]

closed 2023-08-23 by KjellMorgenstern