fritzing/fritzing-app #3952: Error 500 in the Fritzing Discussion Board

BastianFie, 2022-06-08:
I´m sorry for posting this here into the Fritzing GitHub section, however since some days I couldn´t post anything into the Forum. All I get is an error 500:

[image]

I already tried multiple browsers, reboot, deletion from cache, ... could that be fixed?

KjellMorgenstern, 2022-06-08:
Thank you for reporting, I have fixed the forum server, and posts should work again. Also, I updated discourse from 2.8 to 2.9.

closed 2022-06-08 by KjellMorgenstern

BastianFie, 2022-06-13:
> Thank you for reporting, I have fixed the forum server, and posts should work again. Also, I updated discourse from 2.8 to 2.9.

thanks for the quick fix!