fritzing/fritzing-app #3953: Custom sort order for wire colors

tveyben, 2022-06-14:
## Problem
When working in breadboard view and using "wide" jumper cables (like these: https://www.az-delivery.de/en/products/40-stk-jumper-wire-male-to-male-20-zentimeter) it would be really helpfull to have the wire colors listed in the "resistor value order" as that's the order the jumper cables are created in (brown, red, orange, yellow, green ...) currently an other order (for another purpose) is used - and it's not user configurable.

## Proposed Solution
Other issues have been created each suggesting a specific sort order for a given purpose (black + red first as thats used for power etc).

To allow ALL users to be happy a "Wire Color Ordering" option to change the order in preferences should be added.

1. Classic (Default = current order)
2. Resistor Color Code order (1=brown, 2=red, 3=orange, 4=yellow, 5=green ect...)

PR's for even more lists could be added (if other sort orders are logical to implement)

Optionally a user specific list could be implemented, but that might be too much work for the added benefit...!

Observed on Version 0.9.3

KjellMorgenstern, 2022-06-15:
I see zero benefit, but high costs
There would have to be a strong point to make this configurable, which I am missing.
Why would picking a color be any faster or easier if they are in a different order?

Configuration would have to be stored somewhere, a default would have to be defined, other color presets would have to be stored somewhere (I think you description indicates that other presets could not be hardcoded). Errors could sneak in up to messing up colors of existing circuits. The additional feature must be made somehow available, adding complexity to the UI. All that for what?

I agree about showing the colors next to the text in the dialog, and that particular feature would also be easy to implement.

Another idea would be to directly support cables / bundles of wires, in form of a new part. That is a feature which is on our internal roadmap. Implementation should be generic, showing as a new part (or multiple new parts, based on the same implementation), so we could support the mentioned flat-cables, but also USB cables, power cords etc.

tveyben, 2022-06-15:
@KjellMorgenstern What your missing (that I have not clarified) is my use case.
In my case I am mapping a "pre-made" cable with 16 individual wires (in the "resistor color code order" from the breadboard to a 16 relay board, I want to make it easy to select the wire color in the correct order:
1-2-3-4-5-6-7-8-9-0-1-2-3-4-5-6; 1 representing brown, 2 representing red etc
...so I can perform the physical cabling as easy as possible

Currently I have to read each color in the list in order to find my "next wire color" as the order is incorrect for my use case. This is tedious and adds an unnecessary cognitive load (not that it is difficult - it's just extre time consuming).

It's analogue to a "full scan" as opposed to an "indexed lookup" if you relate to databases and similair tasks.

I agree completely with the added code needed in case user specific sort order should be implemented, hence my "optionnally" comment.

I think the **simplest implementation** giving the **most value** would be a setting in the preferences where one can choose between a set of predefined sort orders (resulting in **only** a different order of the values presented in the menu).

This setting **Wire Color Sort Ordering** would have at least to options
* Classic # The current implementation as well as the default value
* Resistor Value # My order based on the resistor color code

I can not identify other sort orders, but that's what I hinted about that if more sort orders were needed, one could proved a new list in a PR

Besides the change in sort order the "Resistor Value" should also prefix the name with the corresponding number as well as the actual color (the latter being common for all listings)
1. Brown
2. Red
3. Orange
etc

KjellMorgenstern, 2022-06-15:
I think the cognitive load ( I am stumbling a bit about that term, it seems nothing compared to implementing this, even this discussion is already worth coloring quite dozens of parallel cables) needs to be done anyhow, since there is now standard for those color cables. They might and the *do* come in variations. Although your cable seems the most common, you would still have to verify that colors are correct. So even if a user knew about the feature - they would not -, how could they trust colors are in their required order?

KjellMorgenstern, 2022-06-15:
Just sure to be understood: I am talking about a new feature, that would allow creating such parts. It is not yet possible to create a cable like that, or at least, it would be very limited in functionality.  The existing format allows arbitrary connections and graphics, so also a 16 or 2 lane cable. But it does not allow to 'behave' like a cable, it would not be flexible in shape and length.

tveyben, 2022-06-15:
OK - makes sense (I mis read you then - thanx for clarifying)

vanepp, 2022-06-23:
I would be in favor of the new cable bundle feature (perhaps with the wire colors specified in the part) as it should provide the desired functionality (at the cost of making a custom part) while providing a lot of other advantages (cleaner looking schematics for one when dealing with bused wires!) and seems a better use of development time.

KjellMorgenstern, 2024-09-30:
@tveyben Reading this back later, I sound quite rude. It was not needed to defend against this issue so strongly, I am sorry.

If I understood correctly, a similar workflow would be:
In a second window, open a sketch that already has the set of wires as desired, for example
[wire_colors.fz.zip](https://github.com/user-attachments/files/17186336/wire_colors.fz.zip)

From the menu "Routing" -> "Select all wires" , press Ctrl+C to copy...
[image]

and Ctrl+V to paste them into your working sketch:
[image]

This should also work within the same sketch, but might be more organized to handle with a second sketch, so the template wires don't get in the way.