fritzing/fritzing-app #3955: Ratsnest colors

KjellMorgenstern, 2022-06-15:
## Current Behaviour
Colors for ratsnest lines seem random, and they change when the ratsnest is recalculated (e.g when distances between parts change)

## Expected Behaviour

Idea: In this order, check
- If we have a wire in the breadboard, the ratsnest line should use the color from that wire (blue most of the time)
- If we have a connection in schematic, the ratsnest line should use the color from that connection (black most of the time)
- If we have a route in the PCB view, the ratsnest line should use the color from that connection (currently hardcoded to 'copper', but might be configurable in future, eg. red and blue for front and back side)

Possible problem:
That will cause all ratsnest lines to have the same color in many cases. That might or might not be a desired effect.

Related problem:
- Make it somewhat easy to sync colors between schematic and breadboard, but the default to blue and black should stay.
-> We could default from black to the breadboard wire color (if any) as soon as the user opens the color picker dialog.

KjellMorgenstern, 2024-01-15:
For example, ratsnest from the schematic would show black, from the PCB would show green, and from the breadboard would show blue? Sounds like that would be simpler (and therfore more realisic) to implement.

Also related ( i think we have such an issue): If there is a netlabel in the schematic, show that label on connections (so, also on ratsnest lines or routed traces)

failiz, 2024-01-15:
> Also related ( i think we have such an issue): If there is a netlabel in the schematic, show that label on connections (so, also on ratsnest lines or routed traces)

That would be great. I don't remember to have seen that issue. I did that in the transitory simulation to name the signals in the oscilloscope. By the way, please, review the animation in the transitory branch. I would like be sure that there are no issues with it.