fritzing/fritzing-app #3956: Custom via sizes don't work anymore in revent versions

dmantione, 2022-06-17:
Hello,

I am not longer able to resize a via by typing the desired dimensions in the properties. For example, I regularily use vias with a hole of 0.3mm and ring of 0.2mm, because this is the minimum size supported by the Chinese PCB manufacturers without additional PCB cost, so for dense PCBs, this is a useful via size.

The last Fritzing version where this works as expected is 0.9.6.

I have created a little screen recording where I show the problem:

https://user-images.githubusercontent.com/7442343/174328283-8810762c-93f1-4a81-976b-166ced75d4ea.mp4

In the video, I am starting with Fritzing 0.9.10 and drop a via on the PCB. Then I try to edit the size by typing the dimensions and fail. Using the dropdown list works, but doesn't give me the required dimensions. I then save the file, quit Fritzing 0.9.10, launch Fritzing 0.9.6 and open the file. I am now able to resize the via as I wish. I save a again, quit, and launch Fritzing 0.9.10. Via still has the right size, but no way to edit it further and once you select another size... you are locked at the wrong size again.

Is this a bug or am I doing something wrong?

failiz, 2022-06-18:
For me works ok in Fritzing 0.9.10 (Windows, Fritzing set to English). 
It may be issue #3080. Could you try to set the language to English?

dmantione, 2022-06-19:
That works! With the language set to English I can adjust the via's.

failiz, 2022-06-19:
Ok, excellent. Then, please close this issue as it is a duplicate. I made a comment in the other issue mentioning that the error also affects the via menu.

dmantione, 2022-06-19:
Thanks, closing bug as requested.

dmantione, 2022-06-19:
Issue is still open despite me clicking the close button... trying again.

KjellMorgenstern, 2022-06-19:
Sidenote: Github App provides only "close as completed" or "wont fix". No "close as duplicate"...

closed 2022-06-19 by KjellMorgenstern