fritzing/fritzing-app #3957: Add ability to create a two row stripboard

KG4RUL, 2022-06-18:
Some stripboards have a two row strip for power distribution.  FRTIZING will only let you create three row strip as the smallest size.

Could the ability to create a two row stripboard be added?

failiz, 2022-06-19:
For reference purposes, could you add a link to some stripboards that have a two row strip for power distribution?

I can confirm the issue. Rows do not allow 2 a valid number. By the way, we should use a standard validator that informs that the number is not accepted (issue: #3954)

Columns can be set to 2. Thus, there is a workaround. Just create a stripboard of 10 rows and 2 columns and rotate the board 90 degrees. After that, you can set the stripboard orientation by creating or cutting strips. If you hold the mouse click, you can hover the mouse over the links for cutting or creating the strips easily.
[image]

KG4RUL, 2022-06-19:
> image

On Amazon, search for ELECTROCOOKIE and you will find many examples of 
their stripboards with 2 row strips for power distribution.