fritzing/fritzing-app #3958: Add shortcut for setting colors to wires

failiz, 2022-06-19:
## Problem
Describes from a user perspective the current state and its shortcomings.

## Proposed Solution
Describes from a user perspective the ideal state of the implementation

KjellMorgenstern, 2022-06-19:
Things to take into considerstion:
- We also have color wires by length.

- Would it help to have shortcut to the context menu? A more generic solution, easier to remember the key/shortcut, less keys reserved, applies to more use cases

- Can we (without much effort) implement automatic coloring based on the connection? E.g. connecting to GND would default to black, Vin would default to red. Parts could contain a section that maps connectors to colors.

failiz, 2022-06-19:
When I am creating small examples, wires by length is not useful as all the wires are small and you want to follow the colour code (black=ground, red=power).

I do not think that the context menu shortcut will be very useful. The time consuming part is the selection of the wire colour menu and then the colour, not the right click to show the context menu.

Automatic colouring would be perfect, bit it is issue #1782. However, I think that implementing the shortcuts could be done much faster. Implementing automatic colouring could take some time. And it will not be perfect. The user will change the colours even in that case as they could want to have a different colour for some wires, specially if there are several voltages (5V=red, 12V=yellow) or for wires that are not power. 

I am not sure if the wire colour for the breadboard view should be part of the part format. If it is, it should be optional. How would you handle cases where the wire connects terminals that should be attached with different colour wires?