fritzing/fritzing-app #3960: Straightening lines in schematic malfunctions.

vanepp, 2022-06-24:
## Current Behaviour

Starting from (remove the trailing.zip to get the .fzz file)

[EC-PH System Wire-PCB Diograme V5.fzz.zip](https://github.com/fritzing/fritzing-app/files/8978906/EC-PH.System.Wire-PCB.Diograme.V5.fzz.zip)

schematic looks like this:

[image]

then move the 2 pull down resistors in order to correct the routing.

[image]

now double click on the join point of the wire (the green arrow)  to straighten it out.

[image]

[image]

then again to straighten the wire more

[image]

here we see the the error, the resistor connection has become unconnected

[image]

The resistor has become disconnected from the net and now has a rats nest line instead of a connection. 

[image]

now establish a connection from the rats nest line

[image]

routing appears correct, all three look to be connected.

[image]

This one is a minor error, but I somehow (which I haven't yet managed to reproduce!) got all three of the pull down resistors connected correctly but not showing a connection in schematic like this:

[image]

where the resistor is in fact connected correctly in the routing database.

[image]

I am working on trying to find the exact steps to reproduce this failure but so far haven't succeeded. I think that both issues may relate to the routing database corruption we sometimes see and thus are probably worth fixing (although this one seems fairly harmless, just reconnect the rats nest lines an all appears well!)

**Build:**

Version 0.9.10 
(bCD-1801-0-29c2cede 2022-05-14) 64 [Qt 5.15.2]

**Operating System:**

Windows 10

**Steps to reproduce:**

see current behavior 

## Expected Behaviour

That the lines straighten and the resistor remains connected to the net as it should.

KjellMorgenstern, 2025-10-08:
Double click on a bendpoint removes it.
Double clicking on a joint (aka bigdot) also removes it.

So, when clicking the bigdot here, it is expected to remove the joint:
[image]

There are several problems:
1. It is not automatically clear that the double click removes the joint. A user could also expect some kind of auto-alignment.
2. In the context menu, the option is "remove bendpoint". This should be labeled differently, e.g. "disconnect joint"
3. It is not clear which connection is going to be disconnected. AB ? AC? BC ? All three?
4. Undo operation is not symmetric (it can not properly be undone)

KjellMorgenstern, 2025-10-08:
Idea:

When hovering over a bigdot, often one wire is highlighted:
[image] 

1. When first hovering over a wire, then moving to the bigdot, that wire should be reliably stay highlighted. Currently, there seems to be some random priority (always the same wire, but not clear how it is chosen)
2. Removing the bigdot detaches the highlighted wire, and places the end in (clear) area about 50px away (adapted to screen resolution, independent from zoom level):

[image]