fritzing/fritzing-app #3962: Improve schematic view in the examples

failiz, 2022-07-16:
## Problem
The examples do not have very good schematics. They do not use power symbols or net labels, which makes the difficult to read. And they do not encourage learners to use them. This will also give more visibility to power symbols and the net labels.

Some examples as an illustration if the issue:
[image]
[image]
[image]
[image]

## Proposed Solution
Improve the schematic view of the examples by using power symbols, net labels, and standard electric engineering practices.
We should also consider if we want to keep wires with different colors in the schematic view. I find that very weird.
See for example my NAND gate schematic (I will add this example to the simulator examples)
[image]

vanepp, 2022-07-20:
"We should also consider if we want to keep wires with different colors in the schematic view. I find that very weird." 

Me too, but some sketches I have seen from users use colors in schematic, so at least some of the user base appears to use them. I'd say if we are considering this we should solicit input from the user community perhaps via the forum (although forum usage has dropped a lot from past years, I don't know of a better place either!)

failiz, 2022-07-21:
I do not think we should remove the functionality, it could sometimes be helpful to illustrate how schematics work. But I think the colors should be removed from the examples.

failiz, 2023-05-30:
The examples of the simulator (and probably all of the others too) are still using the old potentiometers (the pots with the old, and wrong, symbol). And the resistor and the diode still show the "0.25" (max. power of the resistor) and "0.003" (max current of the LED). We should update the parts in the examples to the new versions of the parts to avoid this.

vanepp, 2023-05-31:
Interesting! I haven't really looked at the examples but now that I do I see what you mean. For example the first one I tried: arduino/output/blink has parts in core parts in the temp parts bin (when I think they should be using the ones in core!) 

[image]

I think we would be better off modifying the example .fzz files to use core parts where they exist. I'm not sure we need to do this for the 1.0 release, but we should I think add it to the todo list. Indeed in the simulator example for the pot it is using the old style pot rather than the current one (possibly because we didn't obsolete the old one?) 

[image]

where the one in core parts (this is on 1.0!) is the new pot

[image]

This should be a relatively easy update I expect, just modify the .fzz files in the examples directory.

KjellMorgenstern, 2023-05-31:
@vanepp Are you already using 1.0.0RC2 ? While fixing or cleaning up some of the examples, we had used a dev build with not the official fritzing-parts branch. In release candidate 1 the examples therefore included copies of the parts. For RC2 this should be fixed, but i might have missed one?

vanepp, 2023-06-02:
I think I should be on rc2 on Windows 10

Version 1.0.0 
(rcCD-4149-0-19b5f937 2023-05-25) 64 [Qt 6.4.3]

vanepp, 2023-06-02:
edit2:

Since I'm here :-) 

[image]

has the old pot (I suspect that needs a manual update to get the new one!) another one with a temp parts bin

[image]

old style pot

[image]

old style pot

[image]

old style pot

[image]

old style pot

[image]

old style pot

[image]

old style pot

[image]

old style fsr (I think anyway!) 

[image]

temp parts bin. I think the parts should all be in core (the next one along on perf board uses the same parts with no temp parts bin!) 

[image]

old style pot

[image]

I didn't go through the starter kit or creator kit as AFAIK they are old and probably don't need updating.

KjellMorgenstern, 2024-04-05:
I think these were already solved on 1.0.1 or 1.0.2, however we just checked the examples on a local development build. Therefor I just mark it as solved, to be closed with the 1.0.3 release.

failiz, 2024-04-06:
I do not agree that this issue has been solved. The original issue (bad quality schematic diagrams is still present in most of the examples). I think we have replaced the old pots and force sensor resistors with the new parts, but schematics still do not use GND, power symbols, and net labels and they have wires with different colors.  

See some random examples: 
[image]
[image]
[image]
[image]