fritzing/fritzing-app #3968: Staging and personal repositories

spassignat, 2022-08-16:
Hello,
I create many parts which implies a lot of import and close without saving in Fritzing. Actually working on camdenboss pcb terminals (quite a lot of options).
I remember Fritzing can update its part database from github (it's probably transparent, as I never saw notifications about it). 

It would be great to have a staging repository and a personal repository.

**Staging repository**
- manual update
- automatically import parts
- add and set the subversion or qualifier number. Usually tools use current time millis (gradle, maven, ...) for this purpose.
   - version if the fzp file 1.0.0
   - generated version 1.0.0-9876756

**Personal repository** (a bit like what I understood of the actual mechanism):
- automatic and manual update
- update parts with a newer version

(Having the version number in the file name could help to identify the updated parts, without opening the files)

thanks

KjellMorgenstern, 2022-08-16:
I agree, the current "import, close without saving" cycle for checking parts is bad. I wish for something like selecting the part and pressing "F5".

We disabled that git update feature in the UI, since we were constantly getting errors from users with corrupted parts database. The approach was very error prone. One single conflicting commit had the chance to render all installed Fritzings unusable (this never happened, or at least we detected it fast and were able to force push a correction) 
Hiding the power (or for beginners, complexity) of git didn't yet turn out so well. 

For the same reason, I don't think treating repositories different based on their name is a good idea.
Using milliseconds might be good way to create UUIDs, but is bad to manage versions.

But maybe we can pick up your idea the following way.

Parts could have a flag that marks them as "development". The flag would then cause the following:
- Fritzing would listen to the file system for changes on that parts files
- Big warnings will pester the user when using such a development part in projects with more then e.g. 3 other parts

I am not sure if Fritzing can reload a part without issues if the id is the same. Triggering the obsoletion mechanism for reloading a part was not build to work this way, so there are some risks to this implementation. An alternative could be to auto-increment the part id somehow, and load it as a new part. This approach isn't elegant, but already works if you increment the id manually.

spassignat, 2022-08-16:
If I had to design this feature and considering Fritzing already knows how to properly update parts when their version change, I would look at an existing solution like maven, npm, gradle, nuget ... allowing to publish and manage updates of artifacts. Choose one convenient to integrate (As Fritzing is C++ application, it might worth to have a look at npm). Looking at the repository side (github repository) only may give more work (or not).

Then in Fritzing we would need:
 - a screen to manage artifact repositories
 - a screen to search parts in these repositories (ex: library manager in arduino IDE)
 - a screen to specify which version to use:
   - Resistor v1.0 or 1.0.1 ...
 - some commands to update parts (ex: F5)
 - adjust auto update rules:
   -  (example) only update a part in a sketch to the latest subversion: if the part in the sketch is specified with version 1.0.0, the auto-update would be updated at best to 1.0.x but not to 1.1)
 - a screen to display out-dated parts (the ones having updates)
 - a command to freeze part versions in a sketch (to prevent updates when your pcb is correct, you may not want to take the risk of an update).

With this approach, we would benefit of:
- robust publish/update artifact mechanism
- multiple repositories (official, corporate, personal, staging, ...)
- web interfaces to search, publish... (some repository like nexus, github offer this) 
- the local repository (local file cache) would entirely be manage by this solution.
- an open repository of parts.
- security/permission to access, publish and update artifacts
- license 

What we already have:
- update sketch with new version of parts
- several type of artifacts: bins, parts
- artifact format: zip file + metadatas

KjellMorgenstern, 2022-08-16:
Lets stick to your original use case please, otherwise this ticket will be way out of scope fast. Which i think was reloading a part when it was changed during developing it.

KjellMorgenstern, 2023-01-20:
@spassignat Can you check my latest comment to #3907 ?
I tend to close this #3968 in favor of #3907.
Once implemented, we might tackle some other actionable issues mentioned here.

spassignat, 2023-01-23:
Hi @KjellMorgenstern 

This feature request or suggestion, is to update Fritzing parts database from an external repositories managing the versioning (and publication) of parts:
- either central (inspired from maven, npm, apt, yum, ...) where we can publicly share/download parts
- or local repositories that "advanced" user or small business can use to share their home made parts, which can help to collaborate, make sure every project use the right part version...

There is an overlap with #3907 but only on the update process of Fritzing App local repository from a file, but it does not cover the update of parts from a remote repositories. No need to reinvent the wheel for the repository. There are plenty of existing technologies (npm, Maven, apt, ...) and several backends (ex: sonatype/nexus).

I would say lets keep this one, if this new feature makes sense to everyone.

KjellMorgenstern, 2023-01-23:
From your last description, I think we can close this as already implemented.

Fritzing indeed uses git to manage parts. This was one of the main causes for issues, users would end up with a corrupted part database way too often. Also, the part repo is updated in sync with Fritzing anyway, so we disabled this. Note you can still manually update or switch branches. As I said, this frequently messed up, and we are now deploying Fritzing in a more stable configuration. But you can still start Fritzing with a custom parts directory, which you can then of course manage via git, or any other versioning system of your choice.

closed 2023-11-03 by KjellMorgenstern