fritzing/fritzing-app #3970: Arduino pcb image improvement

spassignat, 2022-08-21:
## Problem
When I layout a card, I always need to look at arduino pinout to know where are the usb,usart connector.

## Proposed Solution
Could it be possible to enhance the pcb image adding a symbol for these connectors ? (simple square is fine)

thanks

KjellMorgenstern, 2022-08-21:
Something like this? 
[image]

spassignat, 2022-08-21:
I was thinking more about something like this:
[image]

KjellMorgenstern, 2022-08-22:
I don't understand, why don't you add the labels then?

spassignat, 2022-08-23:
Hello,
You're right. It should move this to fritzing-part repo. I'm not sure I can move cases from repo to repo. 
Can you do this, for me ?

thanks

closed 2022-08-23 by spassignat