fritzing/fritzing-app #3975: Board cutout should work

spassignat, 2022-09-12:
Hello,

I had some issues exporting a new board for production to gerber format. My cutouts where not exported, while they were coded like the ones I made for maybe 10 other boards.

Today I had the error message, during export:

> "Fritzing is unable to process the cutouts in this custom PCB shape..."

Looking at the code, this error message is produced because a 'z/Z' command is followed by a move 'm/M' command. But in some cases 'z' 'm' sequence is not a problem but in other cases, it is.

Non working code:
```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<svg id="svg24" height="84mm" version="1.2" viewBox="0 0 152 84" width="152mm" x="0" xmlns="http://www.w3.org/2000/svg" xmlns:svg="http://www.w3.org/2000/svg" y="0">
	<g id="board">
		<path id="shape"
			  d="
			  M 16.75,8.5
			  a 1.25,1.25 0 1,0 2.5,0
			  a 1.25,1.25 0 1,0 -2.5,0 z
M 16.75,73.5
			  a 1.25,1.25 0 0,0 2.5,0
			  a 1.25,1.25 0 1,0 -2.5,0 z
M 132.75,8.5
			  a 1.25,1.25 0 0,0 2.5,0
			  a 1.25,1.25 0 1,0 -2.5,0 z
M 132.75,73.5
			  a 1.25,1.25 0 0,0 2.5,0
			  a 1.25,1.25 0 1,0 -2.5,0 z
M 7.25,27
			  a 1.25,1.25 0 0,0 2.5,0
			  a 1.25,1.25 0 1,0 -2.5,0 z
M 7.25,57
			  a 1.25,1.25 0 0,0 2.5,0
			  a 1.25,1.25 0 1,0 -2.5,0 z
M 142.25,27
			  a 1.25,1.25 0 0,0 2.5,0
			  a 1.25,1.25 0 1,0 -2.5,0 z
M 142.25,57
			  a 1.25,1.25 0 0,0 2.5,0
			  a 1.25,1.25 0 1,0 -2.5,0 z
			  Z

			  M 13,1

l 126,0
			  a 2,2 0 0,1 2,2
			  l0,13
			  a 4,4 0 0,0 4,4
			  l4,0
			  a 2,2 0 0,1 2,2
l 0,40

			  a 2,2 0 0,1 -2,2
			  l-4,0
			  a 4,4 0 0,0 -4,4
			  l0,13
			  a 2,2 0 0,1 -2,2
l -126,0

			  a 2,2 0 0,1 -2,-2
			  l0,-13
			  a 4,4 0 0,0 -4,-4
			  l-4,0
			  a 2,2 0 0,1 -2,-2
l 0,-40

			  a 2,2 0 0,1 2,-2
			  l4,0
			  a 4,4 0 0,0 4,-4
			  l0,-13
			  a 2,2 0 0,1 2,-2
Z
"  fill="#1F7A34"
			  stroke-width="0"
			  style="fill:none;stroke:#000000;stroke-opacity:1;stroke-width:0.5;stroke-dasharray:none"/>
	</g>
	<g id="silkscreen">
		<g id="keepout" style="stroke:none;stroke-opacity:1">
			<circle id="l1" cx="18" cy="8.5" fill="none" r="1.5" stroke="#000000" stroke-width="0.1" style="stroke:none;stroke-opacity:1"/>
			<circle id="l2" cx="134" cy="8.5" fill="none" r="1.5" stroke="#000000" stroke-width="0.1" style="stroke:none;stroke-opacity:1"/>
			<circle id="l3" cx="134" cy="73.5" fill="none" r="1.5" stroke="#000000" stroke-width="0.1" style="stroke:none;stroke-opacity:1"/>
			<circle id="l4" cx="18" cy="73.5" fill="none" r="1.5" stroke="#000000" stroke-width="0.1" style="stroke:none;stroke-opacity:1"/>
			<circle id="L1" cx="8.5" cy="27" fill="none" r="1.5" stroke="#000000" stroke-width="0.1" style="stroke:none;stroke-opacity:1"/>
			<circle id="L2" cx="143.5" cy="27" fill="none" r="1.5" stroke="#000000" stroke-width="0.1" style="stroke:none;stroke-opacity:1"/>
			<circle id="L3" cx="143.5" cy="57" fill="none" r="1.5" stroke="#000000" stroke-width="0.1" style="stroke:none;stroke-opacity:1"/>
			<circle id="L4" cx="8.5" cy="57" fill="none" r="1.5" stroke="#000000" stroke-width="0.1" style="stroke:none;stroke-opacity:1"/>
		</g>
	</g>
</svg>
```

Working code:
```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<svg id="svg24" height="84mm" version="1.2" viewBox="0 0 152 84" width="152mm" x="0" xmlns="http://www.w3.org/2000/svg" xmlns:svg="http://www.w3.org/2000/svg" y="0">
	<g id="board">
		<path id="shape"
			  d="
			  M 16.75,8.5
			  a 1.25,1.25 0 1,0 2.5,0
			  a 1.25,1.25 0 1,0 -2.5,0 z
M 16.75,73.5
			  a 1.25,1.25 0 0,0 2.5,0
			  a 1.25,1.25 0 1,0 -2.5,0 z
M 132.75,8.5
			  a 1.25,1.25 0 0,0 2.5,0
			  a 1.25,1.25 0 1,0 -2.5,0 z
M 132.75,73.5
			  a 1.25,1.25 0 0,0 2.5,0
			  a 1.25,1.25 0 1,0 -2.5,0 z
M 7.25,27
			  a 1.25,1.25 0 0,0 2.5,0
			  a 1.25,1.25 0 1,0 -2.5,0 z
M 7.25,57
			  a 1.25,1.25 0 0,0 2.5,0
			  a 1.25,1.25 0 1,0 -2.5,0 z
M 142.25,27
			  a 1.25,1.25 0 0,0 2.5,0
			  a 1.25,1.25 0 1,0 -2.5,0 z
M 142.25,57
			  a 1.25,1.25 0 0,0 2.5,0
			  a 1.25,1.25 0 1,0 -2.5,0 z

			  M 13,1

l 126,0
			  a 2,2 0 0,1 2,2
			  l0,13
			  a 4,4 0 0,0 4,4
			  l4,0
			  a 2,2 0 0,1 2,2
l 0,40

			  a 2,2 0 0,1 -2,2
			  l-4,0
			  a 4,4 0 0,0 -4,4
			  l0,13
			  a 2,2 0 0,1 -2,2
l -126,0

			  a 2,2 0 0,1 -2,-2
			  l0,-13
			  a 4,4 0 0,0 -4,-4
			  l-4,0
			  a 2,2 0 0,1 -2,-2
l 0,-40

			  a 2,2 0 0,1 2,-2
			  l4,0
			  a 4,4 0 0,0 4,-4
			  l0,-13
			  a 2,2 0 0,1 2,-2
Z
"  fill="#1F7A34"
			  stroke-width="0"
			  style="fill:none;stroke:#000000;stroke-opacity:1;stroke-width:0.5;stroke-dasharray:none"/>
	</g>
	<g id="silkscreen">
		<g id="keepout" style="stroke:none;stroke-opacity:1">
			<circle id="l1" cx="18" cy="8.5" fill="none" r="1.5" stroke="#000000" stroke-width="0.1" style="stroke:none;stroke-opacity:1"/>
			<circle id="l2" cx="134" cy="8.5" fill="none" r="1.5" stroke="#000000" stroke-width="0.1" style="stroke:none;stroke-opacity:1"/>
			<circle id="l3" cx="134" cy="73.5" fill="none" r="1.5" stroke="#000000" stroke-width="0.1" style="stroke:none;stroke-opacity:1"/>
			<circle id="l4" cx="18" cy="73.5" fill="none" r="1.5" stroke="#000000" stroke-width="0.1" style="stroke:none;stroke-opacity:1"/>
			<circle id="L1" cx="8.5" cy="27" fill="none" r="1.5" stroke="#000000" stroke-width="0.1" style="stroke:none;stroke-opacity:1"/>
			<circle id="L2" cx="143.5" cy="27" fill="none" r="1.5" stroke="#000000" stroke-width="0.1" style="stroke:none;stroke-opacity:1"/>
			<circle id="L3" cx="143.5" cy="57" fill="none" r="1.5" stroke="#000000" stroke-width="0.1" style="stroke:none;stroke-opacity:1"/>
			<circle id="L4" cx="8.5" cy="57" fill="none" r="1.5" stroke="#000000" stroke-width="0.1" style="stroke:none;stroke-opacity:1"/>
		</g>
	</g>
</svg>

```

The difference is (in the middle):
Z M 13,1 => does not work
M13,1 => works

So I think there is a little issue in the svg path parsing, but I can't find what. It would be better to debug, but I still can't make the project :cry:

So I let you have a look.
Stephane

KjellMorgenstern, 2022-09-13:
Can reproduce this in my dev env. 

To me it looks like the issue is not
"Z M" vs "M" , but "z Z M" vs "z M"

Or in other words, the problem is triggered by two "z" commands in  a row?

spassignat, 2022-09-19:
You're probably right. I think in this case, it's easier to see with a debugger.

KjellMorgenstern, 2023-08-28:
@spassignat Can you create a fzz for each contour? Two 'z' should not be a problem, actually that might already be fixed in Fritzing 1.0.0.  I recently spend a lot of time investigating why my cutout would not work, and it turned out there was no "z" at all, just an "m" that indicated the next path. And most viewers seem to implicitly close the path when they see a new "m", I am not sure if that is according to the specification.

closed 2024-01-08 by KjellMorgenstern