fritzing/fritzing-app #3976: svg scale -1 not fully supported

spassignat, 2022-09-14:
Hello,
I made a part using scale(-1,1) in svg pcb file, to do a mirror. In the graphical screen it's properly displayed, but there is an issue. The DRC fails with overlapping things.

I did an export to Gerber and then I saw my part is displayed in a place but considered at another. (It's a 12 pins header.)
Here I fixed the copper layers but not the silkscreen.
[image]

More or less the same area in Fritzing. As you can see the silkscreen visually appears at the right place. 
[image]

So visually Fritzing support scale(-1) but internally it does not.

KjellMorgenstern, 2023-01-20:
Do you have a minimal fritzing file, like a project that contains such a part, that you can share?

KjellMorgenstern, 2023-04-17:
@spassignat Did you upload the part anywhere? If I just apply scale(-1, 1) to some mock up, it would take more time, and still be somewhat speculative if I have really covered the real world use case.