fritzing/fritzing-app #3977: [LED example] An error occured when starting the simulation

SergioArrighi, 2022-09-15:
## Current Behaviour
Hello, I just opened the led example of the simulator set of examples.
Pushing the simulate button results in an error message:
"An error occurred when starting the simulation"

I downloaded the software today and I'm unsure where to see version number
[image]

**Build:**
0.9.10
<e.g. Version 0.9.3 (b5c895d32 2016-04-19) Cocoa [Qt 5.12.2]

**Operating System:**
MacOS 11.6

**Steps to reproduce:**
- Open simulator basic LED example
- Press simulate

## Expected Behaviour
The simulation starts

failiz, 2022-09-15:
It seems that the ngspice library is not found or that it fails to get the wrapping functions. Did you compile it or is the downloaded from the website?

SergioArrighi, 2022-09-15:
Hello, thanks for reply! I downloaded the installed .dmg

failiz, 2022-09-15:
@SergioArrighi , and did you use the default options or did you modify something during installation? Maybe you could try to uninstall it and try to install it again.
@KjellMorgenstern , did you test the simulation in macos 11.6? Any ideas?

SergioArrighi, 2022-09-16:
Hello,
I simply drag the application icon in the dmg to the application folder, then I activate the simulator in the beta tab.
I tried to delete app and reinstall but I'm experiencing the same issue.
I will try to build it from git repo and let you know.
Many thanks!

KjellMorgenstern, 2022-09-16:
I have tested the 0.9.10 dmg on Big Sur and Monterey. @SergioArrighi report seems plausible, but to be sure: The version information can be seen in the About Dialog. On macOS, you'll find that in the top left menu bar "Fritzing -> About Fritzing". Please send a screenshot.

SergioArrighi, 2022-09-17:
Hello,
sorry I couldn't find it for my initial report.
[image]
Thanks and best regards

lukaslabryszewski, 2022-10-06:
Same issue. Installed the app from the official website DMG file. 
Fritzing Version 0.9.10 (bCD-2134-0-40d23c29 2022-07-01) Cocoa [Qt 5.15.2]
macOS Big Sur Version 11.7

Debug Log:

```
Try path /Applications/Fritzing.app/Contents/MacOS/ngspice.dylib
Couldn't load ngspice Cannot load library /Applications/Fritzing.app/Contents/MacOS/ngspice.dylib: (dlopen(/Applications/Fritzing.app/Contents/MacOS/ngspice.dylib, 5): Library not loaded: /usr/local/opt/gcc/lib/gcc/11/libstdc++.6.dylib
  Referenced from: /Applications/Fritzing.app/Contents/MacOS/ngspice.dylib
  Reason: image not found)
```

offobject, 2022-11-02:
Same issue, just purchased

macOS Mojave 10.14.6
Fritzing 0.9.10

meierich, 2022-11-19:
Same issue 
macOS Catalina 10.15.7
Fritzing 0.9.10
Way - it makes me mad!
Thanks for hints
Erich

meierich, 2022-11-19:
[Andres Faiña](https://blog.fritzing.org/author/andres-faina/)

Dos enyone know to contact Andres Faina ?  (vgl screen shot)

KjellMorgenstern, 2022-11-19:
Guys, dont' be mad, it is clearly marked as beta feature. "Same issue" is a bit thin: For more than half of the issues reported, people had a high value resistor in the circuit, so the LED would not light up. 
Supporting old versions of macOS (which are not even supported by Apple anymore) also is quite a challenge.

failiz, 2022-11-19:
> [Andres Faiña](https://blog.fritzing.org/author/andres-faina/)
> 
> Dos enyone know to contact Andres Faina ? (vgl screen shot)

I'm here.

meierich, 2022-11-19:
Okei - Thanks a lot!
old version of macOS, you got the point.
high value resistor: I was just trying a standard (the simplest) example suggested of Fritzing
regards erich

failiz, 2022-11-19:
> For more than half of the issues reported, people had a high value resistor in the circuit, so the LED would not light up. 

The PR of simulator improvements makes the light of the LED more noticeable at low currents. Are you receiving a lot of issues by mail? I have not seen that reported here.

KjellMorgenstern, 2022-11-19:
Through all kind of channels, as most users are not developers who have a github account. The simple_led.fzz example has a 220 Ohm resistor, but the battery only supplies 3V... so a LED that stays dark is not unusual. The text box next to it suggests to play around with the resistor value. I thought that was intentional, so there is something to discover besides just clicking "Start". Just mentioning it here  so people don't dig into debug logs when the solution is just reading the instructions.

Back to the main issue, the error loading the library. My guess is that it is related to libstdc++.6.dylib as also reported by @lukaslabryszewski . However, it is not yet clear why this happens. The library is included with Fritzing.
In case of Catalina and before, it might be necessary to build against older libraries. But we also got reports on Big Sur and later.

lukaslabryszewski, 2022-11-19:
@KjellMorgenstern 
I'm a software developer, not familiar with inner workings of this application. If there is anything I can do to help you debug this issue please let me know any steps that I could perform to figure this out. Yes, I see the library in the package so not sure why it can't find it. Perhaps a permission or executable signing issue?

lukaslabryszewski, 2022-11-19:
Not sure if this helps:
[image]
[image]

failiz, 2022-11-20:
> Through all kind of channels, as most users are not developers who have a github account. The simple_led.fzz example has a 220 Ohm resistor, but the battery only supplies 3V... so a LED that stays [...]

It was intentional, but maybe it is not the best for the simplest example (I do not use the examples in my teaching, I build the circuits "live".)

If there are too many persons who think the simulator is not working, maybe it is better to decrease the resistance to 100 ohms to make the light noticeable (specially with the PR improvement).

solaSell, 2023-01-25:
Same problem today  January 25 2023. Even the other examples under simulator give similar error message

KjellMorgenstern, 2023-01-25:
@solaSell Good to know you have seen something on January 25 2023 ;-) But it would be more helpful if you added exact Fritzing version, macOS version and error messages (exact as in "screenshot"-exact)

hanskristianjonas, 2023-03-21:
Same problem today (March 21 2023), with pictures.

[image]

[image]

[image]

I have no idea what I am doing, please keep this under consideration when offering help. 
Any help is very much appreciated though.

dd121, 2023-04-03:
Same error:
[image]

[image]

amyeguo, 2023-06-23:
**Using version 0.9.10 of Fritzing, macOS Big Sur Version 11.7.7** 

Experiencing an error when trying to use the simulator, even after decreasing the resistance value from 220 ohms to 100.  
[image]

Here is the debug log when I press "Simulate": 
[image]

failiz, 2023-06-23:
From https://fritzing.org/releases/1-0-0
"Known issues: The simulator does not work on Catalina and Big Sur. A workaround is planned."

KjellMorgenstern, 2024-07-15:
Support for Catalina has been dropped, but the issue remains for Big Sur.

There is still the problem with Big Sur that prevents signed dynamic libraries (ngspice in this case) to load system libraries.

The error message is now clearer:
[image]

"File system relative paths not allowed in hardened programs"

Which is irritating, since a) /usr/local/opt/gcc/... is an absolute path
b) there is no such path in a vanilla Big Sur installation.

If I manually add a copy (from xcode 12)  there, it will complain:
[image]

failiz, 2026-08-18:
It seems this was fixed in #4318

KjellMorgenstern, 2026-08-18:
Well the error pattern shifted over time... but right now we don't have any know issues starting the simulation. Closing.

closed 2026-08-18 by KjellMorgenstern