fritzing/fritzing-app #3978: Curved board shape exported as a list of line

spassignat, 2022-09-15:
Hello,

I created the board bellow.
[image]

Rounded angles are made in svg with arc command, like this:
	  M16.75,8.5
			   a 1.25,1.25 0 1,0 2.5,0
			   a 1.25,1.25 0 1,0 -2.5,0 z

When it's exported in gerber format, the gerber command to draw arcs (seems to be) a long list of short straight lines:

X707Y3022D02*
X708Y3022D01*
X709Y3022D01*
X710Y3021D01*
X711Y3021D01*
X712Y3021D01*
X713Y3021D01*
X714Y3021D01*
X715Y3021D01*
X716Y3021D01*

I don't know the gerber format, but it seems there is a kind of equivalent arc command.
https://www.ucamco.com/files/downloads/file_en/416/the-gerber-layer-format-specification-revision-2021-02_en.pdf?1109f5f3efc7ef822119bd893d685572

The issue is this behaviour raises warning at Aisler manufacturing. I guess those 14000 small cuts might also slow down a lot the manufacturing process.

This enhancement issue need some more investigation on Gerber format to confirm there is a suitable arc command, and to check my analysis is correct.
If there is no arc command, maybe we can add a setting to define the length on these segments.

thanks

failiz, 2022-09-15:
It seems a duplicate of #3185 and #1457. If you think that it is a different bug, please, explain the differences.
If it is the same, it is better to close it to avoid duplicates in the database. You can also give a thumbs up or vote for the issue( if I remember correctly) to indicate that is important to you.

spassignat, 2022-09-16:
Yes maybe the #1457 if I understand well.
Here is an example for gerber conversion for this shape (if it helps):
[image]
SVG
```
<g>
		<path d="M10,10 L90,10" fill="none" stroke="#000000" stroke-width="1"/>
		<path d="M100,20 L100,50" fill="none" stroke="#000000" stroke-width="1"/>
		<path d="M10,50 L100,50" fill="none" stroke="#000000" stroke-width="1"/>
		<path d="M10,10 L10,50" fill="none" stroke="#000000" stroke-width="1"/>
		<path d="M90,10 A10,10 0,0,1 100,20" fill="none" stroke="#000000" stroke-width="1"/>
	</g>
```

Gerber
```
D10*
X10000000Y-50000000D02*
X10000000Y-10000000D01*
X100000000Y-50000000D02*
X10000000Y-50000000D01*
X100000000Y-20000000D02*
X100000000Y-50000000D01*
X100000000Y-20000000D02*
G75*
G03*
X90000000Y-10000000I-10000000J0D01*
G01*
X10000000Y-10000000D02*
X90000000Y-10000000D01*
M02*
`

KjellMorgenstern, 2023-01-20:
We have some improvements for 1.0.0 to reduce "gerber rasterization". Priority right now is to get 1.0.0 stable, setting this to "next" (we definitely want to improve gerber quality)