fritzing/fritzing-app #3980: Hole Position

spassignat, 2022-09-18:
When you have a box with fixed hole position, it's very complex to set the hole position in Fritzing.

Could it be possible to set the position of the center of the hole ? Actually it's the top left corner of the hole part which is positioned.

thanks
stephane

KjellMorgenstern, 2022-09-19:
Not sure why it was done this way in the first place, using the center of the hole for its position would be quite natural.
A fix needs to take into account at least:
- existing projects
- align to grid
- how about vias
- wires connecting to vias

It might be possible to code this in the UI (by adding the rx and ry to the coordinates), which then should not affect exiting projects and positioning helpers like the 'align to grid'

spassignat, 2022-09-19:
Via approach is great but:
- actually it's not really possible to choose the size with no ring and size also is complex to set
- we need to make sure there is no copper surfacing in the hole if there is no ring
    => a check box would be great (better than playing with dimension values)

Being able to connect such via-hole to a wire can be useful for some.

Backward compatibility: have a new hole part (new partId)
Align to grid: use a connector at the center (like via)

"by adding the rx and ry to the coordinates" that makes it flexible, but that can make the code a bit more complex. An 'if' is not a big deal, but you'll have to add some in many places (screen, export modules, save, ...).
The important thing, is rx ry set by the user are very important.

For existing holes or via, their location must not move.