fritzing/fritzing-app #3981: Silkscreen0 on parts

spassignat, 2022-09-19:
## Problem
I made a part which needs to have a silkscreen on both sides.

## Proposed Solution
Could it be possible to have the silkscreen0 available on boards, in the parts ?

thanks

KjellMorgenstern, 2022-09-19:
That should work fine in all recent versions.
But maybe the gotcha is that silkscreen top layer has to be named "silkscreen" not "silkscreen1" ?
See the definitions in the code:

```
...
	        names.insert(ViewLayer::Board,  new NamePair("board", QObject::tr("Board")));
		names.insert(ViewLayer::Silkscreen1,  new NamePair("silkscreen", QObject::tr("Silkscreen Top")));			// really should be silkscreen1
		names.insert(ViewLayer::Silkscreen1Label,  new NamePair("silkscreenLabel", QObject::tr("Silkscreen Top (Part Labels)")));
		names.insert(ViewLayer::GroundPlane0,  new NamePair("groundplane", QObject::tr("Copper Fill Bottom")));
		names.insert(ViewLayer::Copper0,  new NamePair("copper0", QObject::tr("Copper Bottom")));
		names.insert(ViewLayer::Copper0Trace,  new NamePair("copper0trace", QObject::tr("Copper Bottom Trace")));
		names.insert(ViewLayer::GroundPlane1,  new NamePair("groundplane1", QObject::tr("Copper Fill Top")));
		names.insert(ViewLayer::Copper1,  new NamePair("copper1", QObject::tr("Copper Top")));
		names.insert(ViewLayer::Copper1Trace,  new NamePair("copper1trace", QObject::tr("Copper Top Trace")));
		names.insert(ViewLayer::PcbRatsnest, new NamePair("ratsnest", QObject::tr("Ratsnest")));
		names.insert(ViewLayer::Silkscreen0,  new NamePair("silkscreen0", QObject::tr("Silkscreen Bottom")));
		names.insert(ViewLayer::Silkscreen0Label,  new NamePair("silkscreen0Label", QObject::tr("Silkscreen Bottom (Part Labels)")));
...
```

KjellMorgenstern, 2023-01-20:
@spassignat I guess it worked by my suggestion above. Please reopen if it doesn't or you exprienced a bug trying to add a silkscreen layer.

closed 2023-01-20 by KjellMorgenstern