fritzing/fritzing-app #3982: Option to lay out PCB the same as the breadboard, instead of randomly

gilesp1729, 2022-10-03:
## Problem
You're building a breadboard layout, and you want to move to a PCB. Presently the parts are randomly scattered on the PCB layout, and it is difficult (almost impossible) to find them all and reposition them before you can even begin optimising the PCB. 

## Proposed Solution
If you're starting out by placing them on a breadboard you have a physical layout and should be able to tell the PCB layout to follow it. It might not be the best for a PCB, but it's a lot better starting point for a PCB than a random layout. The PCB layout could automatically follow the breadboard layout, until you start routing traces and moving things around on the PCB. And/or you could do it manually.

failiz, 2022-10-03:
Related and very similar to #1535
I am not in favor of doing this by default as it could cause problems to other users. Maybe a button to rearrange the parts based on some view could be useful in some occasions, but it should be called every time you want to rearrange the components. If so, it should be generic: You could trigger it from any view to follow the arrangement of another view.

KjellMorgenstern, 2022-10-04:
Note that recent version of Fritzing already try to do this for the schematics view. It works more or less, but once you move parts around in one of the views, syncing is no longer possible. And also not desirable in most cases.

gilesp1729, 2022-10-20:
I will try to copy the coordinates from the breadboard view into the PCB view using an external program to parse out the XML in the .fz file. It looks fairly straightforward. That would give a PCB the same size as the breadboard with the parts in the same relative positions, and a simple single-sided routing should be possible. Optimisation by hand from there of course. It has to be better than a whole lot of random positions that need to be undone by hand.

gilesp1729, 2022-10-23:
Results of doing the copy of coordinates as per last reply. A work in progress - stay tuned!
Here is the breadboard:
[image]
The PCB before the copy:
[image]
The PCB after the copy. Not too bad, but there are some issues with the origin coordinates of some parts being different in each view. In particular, the IC seems to be a bit confused about where its centre is. Or there could just be a bug in my program.
[image]

KjellMorgenstern, 2023-01-20:
@gilesp1729 How happy are you with your result? 

To work the best, Fritzing parts would need a standardized reference point and orientation, but I guess the feature might be useful even with that limitations.