fritzing/fritzing-app #3983: Simulation error

Zenande-dev, 2022-10-19:
Hey

I experiencing simulation error and I don’t know how to solve it. Can you please assist
[image]

failiz, 2022-10-19:
It seems that there is no model for the DI_1N4732A. Is this a custom part or a core part? If you can, you could also upload the Fritzing file.

Zenande-dev, 2022-10-19:
[Fritzing.zip](https://github.com/fritzing/fritzing-app/files/9819804/Fritzing.zip)

failiz, 2022-10-19:
Thanks for reporting the bug. It is a core part (drop diode from core bin and change it to zener). 

The problem is in the spice field of the part. It is using a subcircuit and the spice line needs to start with X instead with D. It will be fixed in the next release. Until then, if you need a zener diode, you can create a new part.

In any case, remember that the simulator cannot simulate Arduinos. See its capabilities here: https://blog.fritzing.org/2022/06/27/Simulating-Circuits-with-Fritzing

failiz, 2022-10-19:
The pull request that fixes the issue: https://github.com/fritzing/fritzing-parts/pull/351

KjellMorgenstern, 2023-01-20:
PR was merged, marking as "solved" unitl we can close this with the 1.0.0 release.

closed 2023-08-13 by KjellMorgenstern