fritzing/fritzing-app #3985: Capital Z in SVG path prevents cutout detection

dmantione, 2022-11-04:
If you load the following SVG file as PCB shape:

```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<svg width="100mm" height="100mm" viewBox="0 0 100 100" version="1.1">
  <g id="board">
    <path fill-rule="evenodd" d="M 0,0 H 100 V 100 H 0 Z
	      M25,25 H 75 V 75 H 25 Z"
	  style="stroke-opacity:1;fill-opacity:1;stroke:#ffbf00;stroke-dasharray:none;stroke-miterlimit:4;stroke-width:0.0;fill:#00b629"
    />
  </g>
  <g id='silkscreen'></g>
  <g id='silkscreen0'></g>
</svg>
```
... no cutout is detected by Fritzing.

However, if you convert Z into z:

```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<svg width="100mm" height="100mm" viewBox="0 0 100 100" version="1.1">
  <g id="board">
    <path fill-rule="evenodd" d="M 0,0 H 100 V 100 H 0 z
	      M25,25 H 75 V 75 H 25 z"
	  style="stroke-opacity:1;fill-opacity:1;stroke:#ffbf00;stroke-dasharray:none;stroke-miterlimit:4;stroke-width:0.0;fill:#00b629"
    />
  </g>
  <g id='silkscreen'></g>
  <g id='silkscreen0'></g>
</svg>
```

... a cutout is detected just fine.

Would it be possible to get this fixed? It is only a minor annoyance when editing SVG files by hand, but as soon as you start designing them with a tool like Inkscape, it becomes quite annoying, since you must correct the file by hand over and over again.

KjellMorgenstern, 2022-11-04:
resizableboard.cpp:
			QStringList ds = element.attribute("d").split("z", Qt::SkipEmptyParts);
->
			QStringList ds = element.attribute("d").split("z", Qt::SkipEmptyParts, Qt::CaseInsensitive);

dmantione, 2022-11-04:
Wow, that was fast, thanks!

KjellMorgenstern, 2022-11-04:
I think the same needs to be done again for the gerber export later. In ```dealWithMultipleContours``` there is a case sensitive split, too. We had a similar issue: #3537 . Somehow it worked for the #3537 example, probably there were only subpaths with a single 'Z' in the example for that issue.

KjellMorgenstern, 2022-11-04:
@dmantione Thanks for the clear report

KjellMorgenstern, 2023-08-23:
There is another issue, if there is no closing z at all, but a the cutout path just starts (with an 'm'). 
This is somewhat an edge case, because the contour would not be a closed path. But Fritzing treats it as a closed path, and keeps silent about the ignored cutout.

closed 2023-08-23 by KjellMorgenstern