fritzing/fritzing-app #3990: QMake error on Ubuntu 18.04: Project ERROR: quazip could not be found.

RobertPHeller, 2022-12-07:
What am I missing?

sauron% /usr/lib/qt5/bin/qmake
Project MESSAGE: found libgit2 include path at /home/heller/fritzing-app/../libgit2/include
Project MESSAGE: found libgit2 library in /home/heller/fritzing-app/../libgit2/build
Project MESSAGE: Using fritzing boost detect script.
Project MESSAGE: using installed Boost library
Project MESSAGE: Using fritzing quazip detect script.
Project ERROR: quazip could not be found.
sauron% dpkg-query -l \*quazip\*|grep ^ii
ii  libquazip5-1:amd64   0.7.3-5ubuntu1 amd64        C++ wrapper for ZIP/UNZIP (Qt5 build)
ii  libquazip5-dev:amd64 0.7.3-5ubuntu1 amd64        C++ wrapper for ZIP/UNZIP (development files, Qt5 build)
ii  libquazip5-headers   0.7.3-5ubuntu1 amd64        C++ wrapper for ZIP/UNZIP (development header files, Qt5 build)

RobertPHeller, 2022-12-07:
I'm trying to build 0.9.10 from the GitHub repo, under Ubuntu 18.04 (amd64).

RobertPHeller, 2022-12-15:
OK, I will try that.

At Thu, 15 Dec 2022 11:49:09 -0800 fritzing/fritzing-app ***@***.***> wrote:

> 
> Two points with the current build system:
> * It is only looking for QuaZIP in a specific directory
> * It is looking for QuaZIP 1.x
> 
> For the first point, I suggest supporting pkg-config #3993 - but looks that will only work with Ubuntu 23.04 onwards when libquazip1-qt5-dev becomes available in Ubuntu:
> https://packages.ubuntu.com/search?keywords=libquazip1-qt5-dev&searchon=names&suite=all&section=all
> 
> To actually try and answer the question. If QuaZIP 0.7 still works with v0.9.10, may be possible to modify pri/quazipdetect.pri to find the 18.04 version something like (untested) :
> 
> ```diff
> --- a/pri/quazipdetect.pri
> +++ b/pri/quazipdetect.pri
> @@ -5,20 +5,14 @@ message("Using fritzing quazip detect script.")
>  SOURCES += \
>      src/zlibdummy.c \
>  
> -exists($$absolute_path($$PWD/../../quazip_qt5)) {
> -        QUAZIPPATH = $$absolute_path($$PWD/../../quazip_qt5)
> -        message("found quazip in $${QUAZIPPATH}")
> -    } else {
> -        error("quazip could not be found.")
> -    }
> +QUAZIPPATH = /usr
>  
>  message("including $$absolute_path($${QUAZIPPATH}/include/quazip)")
>  
>  unix:!macx {
>      message("including quazip library on linux")
> -    INCLUDEPATH += $$absolute_path($${QUAZIPPATH}/include/quazip)
> -    LIBS += -L$$absolute_path($${QUAZIPPATH}/lib) -lquazip1-qt5
> -    QMAKE_RPATHDIR += $$absolute_path($${QUAZIPPATH}/lib)
> +    INCLUDEPATH += $$absolute_path($${QUAZIPPATH}/include/quazip5)
> +    LIBS += -lquazip5
>  }
>  
>  macx {
> ```
> 
> 

-- 
Robert Heller             -- Cell: 413-658-7953 GV: 978-633-5364
Deepwoods Software        -- Custom Software Services
http://www.deepsoft.com/  -- Linux Administration Services
***@***.***       -- Webhosting Services

closed 2023-01-20 by KjellMorgenstern

Splarkszter, 2023-06-28:
Seems quazip it's required but wasn't on my system, I didn't find it on the installation instructions tho.

I did `$ sudo apt-get install libquazip5-dev` but I still get `quazip not found` error on Linux Mint 21.1