fritzing/fritzing-app #3991: startup crash on ubuntu

sktrinh12, 2022-12-08:
## Current Behaviour
I get the error `qt.gui.icc: Unsupported ICC profile class 70727472
QPngHandler: Failed to parse ICC profile`
upon startup. I just downloaded latest version on my laptop and opened from terminal. The logo shows up loads for about 5 seconds disappears (the logo) and no window appears.
<please attach screenshot if possible>
**Build:**
fritzing-0.9.10-l2134-40d23c29.AppImage

**Operating System:**
Ubuntu 20.04

**Steps to reproduce:**
- Step 1 - download from fritzing website newest linux AppImage
- Step 2 - chmod +x fritzing*.AppImage file
- Step 3 - ./fritzing*.Appimage

## Expected Behaviour
Open fritzing application

Thank you!

KjellMorgenstern, 2022-12-08:
The ICC profile error message is harmless, it is passed through from a library, originating from one PNG image with an invalid/unknown color space.

The reason why no window appears I can not tell. Which window manager are you using?

sktrinh12, 2022-12-09:
i'm using i3 window manager. i tried looking thru my windows and couldn't find it.

KjellMorgenstern, 2023-01-20:
Anyone else with Ubuntu 20.04 and i3 to confirm the issue? Otherwise, this has to wait until I find the time to setup a system.

KjellMorgenstern, 2023-09-17:
There were quite some changes from 0.9.10 to Fritzing 1.0.1 
Closing this for now, please comment if you think this should be opened again.

closed 2023-09-17 by KjellMorgenstern