fritzing/fritzing-app #3996: Metadata editor has no labels

fbitterlich, 2022-12-18:
## Current Behaviour
On a brand-new installation, the "Metadata" tab in the new Parts Editor is not showing labels for the input fields, so it is impossible to known what goes into which field.

See screenshot.

**Build:**
Version 0.9.10 
(bCD-2134-0-40d23c29 2022-07-01) Cocoa [Qt 5.15.2]

**Operating System:**
macOS 11.6.1 on an M1 iMac

**Steps to reproduce:**
- Right-click on a part and select "Edit Part (new parts editor)"
- Click on the "metadata" tab

## Expected Behaviour
- To see a number of fields with labels which explain what the content is supposed to be. The fields are there, the labels not.

[image]

vanepp, 2022-12-28:
As a data point it appears to work fine on 0.9.10 on Win10

[image]

so this looks to be either Mac specific or your configuration specific.

KjellMorgenstern, 2022-12-29:
@vanepp Thank you :-) That means it is very likely that the desktop is configured for dark mode, and the dialog backround switches to dark mode, but the text color stays at a hardcoded value.

fbitterlich, 2023-01-02:
Confirm, it happened in dark mode.

MDBInd, 2023-05-03:
I am experiencing this issue on Ubuntu 22.04 in dark mode and Fritzing Version 0.9.6 (b0.9.6 2021-02-21) 64 [Qt 5.15.3].

When I change the OS to light mode and restart Fritzing, I can see the labels.

MimiVRC, 2024-04-29:
this issue still exists today if your os is set to dark mode on windows 10 as well