fritzing/fritzing-app #4000: "Sorry, we have a problem with the swapping mechanism" on latest commits

cobalt2727, 2023-02-15:
## Current Behaviour
Currently building Fritzing from source using the latest commit in the `develop` branch for both `fritzing-parts` and `fritzing-app`, and trying to run the resulting build throws this error message at me.
[image]
```ls
Version 0.9.9 ( 2021-09-22) 64 # (this is actually just the latest commit)
```
**Operating System:**
<Pop!_OS 22.04>

**Steps to reproduce:**
```bash
#!/bin/bash

sudo rm -rf /tmp/fritzing-app
sudo rm -rf /usr/share/fritzing/fritzing-parts #yes I know this is horrible practice I'm just lazy
echo "WARNING: If you previously had Fritzing installed somewhere, it's now completely broken until this script finishes or you reinstall it from elsewhere."

git clone --branch develop --single-branch https://github.com/fritzing/fritzing-app --depth=1 /tmp/fritzing-app
sudo git clone --branch develop --single-branch https://github.com/fritzing/fritzing-parts /usr/share/fritzing/fritzing-parts --depth=1 #yes I know this is horrible practice I'm just lazy
sudo rm -rf /usr/share/fritzing/fritzing-parts/.git*
cd /tmp/fritzing-app

#if the folder already existed, the 'git clone' step would have failed
#so let's check for updates:
git pull

### the following lines were blatantly stolen from the Arch User Repository package for building Fritzing
### https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=fritzing
# Dynamically link against system libgit2
sed -i 's/LIBGIT_STATIC = true/LIBGIT_STATIC = false/' phoenix.pro
# Disable broken font scaling (#3221)
sed -i 's/Exec=Fritzing/Exec=env QT_AUTO_SCREEN_SCALE_FACTOR=0 Fritzing/' org.fritzing.Fritzing.desktop

#prepare the build
#the name of the command here is different for different systems - if qmake-qt5 fails, fall back on qmake
qmake-qt5 || qmake
#actually build it now
#nproc returns the number of processors; appending it to 'make' like this will multithread the build
make -j$(nproc) && echo "BUILD SUCCEEDED! Installing:"

#install the build
sudo make install
```
## Expected Behaviour
not seeing this error message

cobalt2727, 2023-02-15:
I switched `fritzing-app` to the `main` branch instead of `develop` and now it can't find the system quazip, despite earlier commits/PRs suggesting this should be possible now...

KjellMorgenstern, 2023-02-15:
Can you reproduce the issue with the official build from fritzing.org?

vanepp, 2023-02-28:
For your swapping mechanism error once Fritzing loads click on breadboard (any view other than Welcome will do) then Part->regenerate parts database (which will do the update then exit Fritzing) on restart Fritzing should load correctly.

cobalt2727, 2023-03-01:
Where would I get my hands on `parts.db`?
[image]
Looks like it's not present in the repo:
https://github.com/fritzing/fritzing-parts/search?q=parts.db

vanepp, 2023-03-01:
Looks like you missed this step in the build instructions available here

https://github.com/fritzing/fritzing-app/wiki/1.-Building-Fritzing

...
In Qt Creator, open the Fritzing project by navigating to the Fritzing source folder and select the phoenix.pro file. Then go to the Projects tab, and under Run Settings, add this in the field Arguments:

-f "/path/to/fritzing-app/" -parts "/path/to/fritzing-parts/" -db "/path/to/fritzing-parts/parts.db"

    -f [this flag is required for fritzing to e.g. find the translation-files]
    -parts [find the fritzing parts directory]
    -db [rebuild a clean parts.db from the parts directory]
...

which builds the parts database from the repository. Note I haven't done a source build in several years, so procedures may have changed but I  expect this still applies and is necessary to get a parts database built.

vanepp, 2023-03-30:
I expect there should be (such as running fritzing.exe with those parameters) but I have only ever built from QT Creator so I don't know how to do it from the cli I suspect QT Creator is running Fritzing.exe with the listed parameters but I'm not sure of that (it may be doing something internal to QT as well). Sorry.