fritzing/fritzing-app #4002: It's hard to select objects in Breadboard view, not selecting the object the cursor is pointing at.

djom202, 2023-02-19:
## Current Behaviour

This is happening from I got the app, so I reinstalled multiple times but I still have the same issue and It so hard to try to edit or move the elements since its take a lot of time to find the exactly point into the Breadboard which is the position needed to edit, move, or clicked the elements. It seems the issue start when I add a lot of elements (boads, wires, LCD screens, micro controllers, etc.)

[image]

**Build:**
Version 0.9.10 (bCD-2134-0-40d23c29 2022-07-01) Cocoa [Qt 5.15.2]

**Operating System:**
MacBook Pro 2020 with MacOS 13.1 version

## Expected Behaviour
Hovering the pointer over elements, wires, or junctions should make it selectable.

KjellMorgenstern, 2023-02-19:
Is it possible to share a circuit where this happens?

djom202, 2023-02-23:
Hi @KjellMorgenstern from my site I can't share the diagram, since I going to try to create another one similar that not include sensitive information.

djom202, 2023-02-23:
[ESP32.zip](https://github.com/fritzing/fritzing-app/files/10817779/ESP32.zip)

vanepp, 2023-02-25:
While I'm not Kjell, I'm not seeing a problem here. This is on Windows10 not a Mac, but it appears to be working normally to me.   Here I clicked on the end of the wire (indicated by the red arrow) and moved it to the end of the breadboard (which makes the image much more readable and less error prone!) 

[image]

which results in this (as noted on Windows10) what different happens for you? Can you provide a screen show of what error you are seeing? 

[image]

Ah! I may have discovered what you are seeing. Clicking on the pin of the ESP board like this does not select the attached wire:

[image]

but clicking on the wire does select it

[image]

However on windows either click will cause the wire to be re positioned if I left click and drag the wire. Is that not happening for you?

Peter

closed 2025-07-29 by KjellMorgenstern