fritzing/fritzing-app #4003: Fritzing cannot handle breadboard in inches correctly

VirtualLuthier, 2023-02-24:
Hello,
I am writing a fritzing breadboard generator in python.
Using fritzing 0.9.10 on windows 10
I have created a breadboard in inches (x="0in" y="0in" width="6.6in" height="2.8in" viewBox="0 0 6.6 2.8" enable-background="new 0 0 6.6 2.8"), all svg coordinates are in inches.
It looks perfectly in Inkscape and also fritzing can display it correctly.
BUT when hovering over it, the "small" circles of the contacts are absurdely enlarged. Also, if I place a component on it, the green "bus" contacts are much too huge. Is there any chance, that this is corrected? Or what must I do without spoiling the svg file?

[image]
[BroadBreadBoard.fzp.txt](https://github.com/fritzing/fritzing-app/files/10823425/BroadBreadBoard.fzp.txt)
[image]

vanepp, 2023-02-25:
Your best bet would be to upload the .fzpz file for your breadboard either here (where you will need to append a .zip to the .fzpz) or to the Fritzing Forum. I have made a number of breadboards and can probably tell you what is wrong with yours given the .fzpz file.

Peter

VirtualLuthier, 2023-02-25:
I do not know exactly what has to be in the fzpz file for a breadboard. But as you see, I have already uploaded 
-the svg file
-the fzp file (as .txt file)
Missing was the icon file(now attached) and the xml snippet for the installation. Are they important?
[image]

the snippet:

```
		<instance moduleIdRef="BreadBoardBroadModuleID" modelIndex="49" path="BreadBoardBroad.fzp">
            <views>
                <iconView layer="icon">
                    <geometry z="-1" x="-1" y="-1"/>
                </iconView>
            </views>
        </instance>
```
I hope, this helps

vanepp, 2023-02-26:
Ah! I missed the fact the svg is here. Creating an fzpz is easy. Rename BroadBreadBoard.fzp to part.BroadBreadBoard.fzp, and BroadBreadBoardMain.svg to svg.icon.BroadBreadBoardMain.svg and svg.breadboard.BroadBreadBoardMain.svg then zip the 3 file (the fzp and 2 svgs) to BroadBreadBoard.fzpz and you are away. Indeed that breaks for me too and I don't see why necessarily. My guess is the scale.

[image]

your svg is set for 1 drawing unit = 1inch, I suspect the code wants a much smaller scale (the usual is 1 drawing unit = 1/1000in). I'll rescale one of my breadboards to be a scale of 1.0 and see if it does the same thing.  OK my test breadboard works correctly (at scale 0.001) 

[image]

Now I will rescale the svg to 1.0 and see if that breaks. Yep, that looks to be the problem OK.

[image]

so the solution is fairly simple (if a lot of work!) You need to rescale your svg to 0.001 from 1.0. To do that you need to ungroup it which is where the work comes in. The breadboards have a ugly format where the connectors are groups so if you ungroup them the connectors disappear. Early on I fixed that by moving all the connectors from the group to a drawing element (which come to think of it if I give you that svg properly scaled should do what you need with no particular work!) So lets do that assuming I can figure out how to get an svg here. That is easy, just copy it here. This is the working svg at the correct scale 

[image]

the difference is here, I used a python script to move the connector names from the group to the circle

[image]

so you can ungroup the svg to rescale it and not lose the connectors. As well here are the two fzpz files (just delete the trailing .zip to get back the .fzpz) the moduleIds are the same so you can only load one at a time in to fritzing.

[breadboard2-test.fzpz.zip](https://github.com/fritzing/fritzing-app/files/10833059/breadboard2-test.fzpz.zip)

[breadboard2-rescaled.fzpz.zip](https://github.com/fritzing/fritzing-app/files/10833060/breadboard2-rescaled.fzpz.zip)

Hopefully that gives you enough to figure out what you need to do.

Peter

VirtualLuthier, 2023-02-26:
Thank you for the detailed help.
I have tried now the following:
x="0in" y="0in" width="6.6in" height="2.8in" viewBox="0 0 6600 2800" enable-background="new 0 0 6600 2800" and additionally used the main node:
<g id="breadboardbreadboard" transform="matrix(1000,0,0,1000,0,0)">
		<g id="background">
			<rect x="0" y="0" width="6.6" height="2.8" fill="#d9d9d9"/>
		</g>
And I left all the coordinates in inches as before.
The behaviour is the same as before. I decided, that I do not like to do some coordinate voodoo on ALL of the coordinates, just because fritzing cannot handle the correct values. Fortunately I am not urged to use fritzing.
I will give it another try in one year or so and abandon the project for now. By the way, If I create the same file in mm, the according spots are just a bit too small, but only a bit. I think, fritzing should solve its mm, inch, and dpi coordinate mess. I have seen, that your solution works with a 7.2 distance between adjacent pins. This seems to me, as if the 72dpi setting still plays an important role here.
So my recommendation here: treat this as a bug

KjellMorgenstern, 2023-02-26:
I tried the board, and one issue I found is much simpler.

In your SVG, you don't specify the stroke-width for the circles that are used for the connectors.
Fritzing uses some default stroke width for highlighting connectors, which is not adapted to your boards scale.

At least in my soon to be released development version, adding a default stroke width seems to work fine:
` <g id="breadboardbreadboard" stroke-width="0.001">`

[image]

I didn't try with the current release 0.9.10, maybe you need to add stroke-width to each connector. We have fixed some of the parts about inheriting transformations and properties from parent nodes in SVG.

Of course, this is just a workaround, and Fritzing should take the scale into account for the default value.

VirtualLuthier, 2023-02-26:
Unbelievable!
When giving a stroke-width it really works, with mm and in. 
Thank you very much!
I attach the new mm file with 800 stroke-width's added.
[image]

closed 2023-02-26 by VirtualLuthier