fritzing/fritzing-app #4005: Requesting improvement for making custom PCB shapes

PickyBiker, 2023-02-26:
Making custom PCB shapes for Fritzing is a difficult task, this needs to be made much easier.

I would like to see the ability to import a .svg shape and have Fritzing automatically convert it to a PCB ready to use. If changes are necessary, then just redraw the .svg and import it again.

KjellMorgenstern, 2023-02-26:
This is already pretty easy, you can load a custom shape here:

[image]

PickyBiker, 2023-02-27:
You are correct. I gave a poor description of the problem. It is not the loading of the image; it is the requirements of the .svg image using inkscape that is difficult. I guess it is really an inkscape issue.

KjellMorgenstern, 2023-02-28:
So for example you would like to be able to load an SVG path like that:

https://yqnn.github.io/svg-path-editor/#P=M_10_10_L_100_10_L_55_90_Z_M_40_40_L_70_40_L_55_70_Z

And have it automatically used as a board by Fritzing?

PickyBiker, 2023-02-28:
Yes, that would be great!