fritzing/fritzing-app #4006: Schematic view fonts on terminals too light

raybonz1, 2023-02-27:
## Problem
When printing the schematic you cannot read the terminals because the font is pale grey color and print cannot be read when printed,

## Proposed Solution
Change terminal font color to a bold color such as black for good contrast and better printouts. Maybe an option to change terminal font color in preferences would be even better. Checked preferences and there is no option to change font or color of font.

vanepp, 2023-02-27:
The colors are specified (but not always obeyed in parts!) in the graphics standards here:

https://fritzing.org/fritzings-graphic-standards

so presumably a change to the graphic standards would be needed to change this. Updating core parts would also be a lot of work (and changing user provided parts basically impossible) so even a change may be difficult.

raybonz1, 2023-02-27:
I am referring the terminals on the Uno board and LCD that are pale gray in the attached pic. Printed terminal text is unreadable with this font color. Needs to be a dark color with good contrast. Maybe next software update add font colors to Preferences? Common practice in many Windows apps including 3D modeling programs.
Thanks,
Ray
***@***.***

https://fritzing.org/fritzings-graphic-standards

raybonz1, 2023-02-28:
[image]
Wonder if this image will appear lol..

vanepp, 2023-02-28:
Yes your image appears. A solution to your problem would be to create custom parts for both the UNO and the LCD and change the fill on the pin names. Currently the UNO doesn't meet the graphics standards (this is the schematic svg displayed in Inkscape) 

[image]

the fill is #8c8c8c (not the #555555  specified in the graphic standard)  that looks like this

[image]

setting it to black like this may be what you want and can easily be done in a custom part (assuming you know how to edit svgs in Inkscape or another svg editor.)

[image]

At some point in time we may get around to updating all the parts in core parts to meet the graphics standards but that hasn't occurred yet. Until then a custom part is your best bet. 
This tutorial set in the fritzing forum will tell you how to create custom parts

https://forum.fritzing.org/t/part-creation-howto-part-1-breadboard-and-pcb/7692

it is however reasonably complex. Assistance is available in the forum though.