fritzing/fritzing-app #4009: Reading Voltage Impossible?

shaneparlier, 2023-03-03:
[image]
[image]
# Current Behavior *Initially posted in Fritzing forum*
Unable to read the correct voltage when meter is applied correctly to parts with SPICE simulation. Voltage reads 3vdc on simulated open circuit.
<please attach screenshot if possible>
[image]
[image]

**Build:**
Version 0.9.10
(bCD-2134-0-40d23c29 2022-07-01) 64 [Qt 5.15.2]

**Operating System:**
Windows 11 Pro
22621.1265

**Steps to reproduce:**
- Step 1 Create simple circuit with voltmeter attached at appropriate anode/cathode terminal and test with switch simulated open vs closed.

- Step 2 Simulate, voltage should read 0vdc with open circuit, and 3vdc closed (that's not the case in this situation).

## Expected Behavior
Voltage should read 0vdc with open circuit, and 3vdc closed circuit.

failiz, 2023-03-07:
Thanks for reporting the issue. 
This is due to the spice model of the switch. I used a switch model (see section 3.3.15 and 3.3.16 of the ngspice manual). From the manual:
"The switch model allows an almost ideal switch to be described in ngspice. The switch is not quite ideal, in that the resistance can not change from 0 to infinity, but must always have a finite positive value. By proper selection of the on and off resistances, they can be effectively zero and infinity in comparison to other circuit elements."

In this case, the switch has a resistance of 1m ohm when is closed and 1M ohm when is open. Thus, the simulation is correct but the model is not able to simulate the reality in this case. 

I chose the switch model as a way to allow us to simulate pulsations in a switch when we implement transitory simulations. However, I think we should probably change the model to an open circuit when the switch is released. This will make the simulation correct in these cases. In addition, there are two parts that get swapped (the release and pressed buttons are two parts), so we are not using the switch model of ngspice. We could have a special switch part when we implement transitory analysis. @KjellMorgenstern , do you agree with this change? If so, I will make a pull request.

KjellMorgenstern, 2023-03-09:
So we would exchange the part because it can not be properly modeled in SPICE ? 
This looks like a red flag, I'd like to discuss this a bit more. 
Are there other situations where the SPICE model needs to be exchanged? 
As I understood, the fix would be so that instead of reporting a high resistance, we don't list the connection at all for simulation. And to achieve this, the plan is to switch the part in Fritzing, like if the user removed it and added another part in place. Is this correct?

failiz, 2023-03-09:
Currently, we already have two parts for every switch that can be simulated. This is done to change the spice field, but also to change the images in the bb and sch views (in sch the switch is closed or open and in bb the button is pressed or not). Thus, the change will only change the spice field of the open switches and it will not be necessary to obsolete the part. 

I agree that this technique is not very elegant, but I did not see many other alternatives as I wanted to change the svg of the part based on the status of the switch. The IR sensor FC-51 also uses this tecnique. In the case of the  FC-51, the images are the same and we could implement something more elegant as the only thing that changes is the spice field. However, this would add more code to the program as the spice field will depend on the status property. I could try to be able to create a spice field that acts conditionally to the status field, but not sure how easy it is.

failiz, 2023-04-26:
Commit below fixes this issue:
https://github.com/failiz/fritzing-parts/commit/eaa7f61e40952bc514e82b1321c7e3115617fb88

KjellMorgenstern, 2023-05-17:
Reopening, so we can keep track until the improvement really found its way into Fritzing 1.0.0.

KjellMorgenstern, 2023-05-17:
Switch now works as expected.
This is a slightly different curcuit then in the OP, but I think it shows the same problem. Before the open switch allowed some voltage drop, which disappears with the new model.
[image]

closed 2023-08-24 by KjellMorgenstern