fritzing/fritzing-app #4012: No control over connections made over perfboard

dronus, 2023-03-11:
## Current Behaviour
When in breadbord view, connecting a wire from a part or breakout to another one over perfboard connects the wire from the first part's pin to the breadboard "pin". So parts can't be moved around without breaking about half of the connections.

**Build:**
<Version 0.9.6 (b ) 64 [Qt 5.15.2]

**Operating System:**
<Ubuntu 22.04>

**Steps to reproduce:**
- Switch to breadboard view
- Remove breadbord template
- Add some perfoboard
- Place parts over the perfoboard
- Connect pin of those parts
- Move some parts around

## Expected Behaviour
All connections should connect to the parts first, and move with it if part is rotated / moved. Only use the perfoboard "pin" if wires are to be connected to each other.

dronus, 2023-04-10:
When I connect a wire _from_ an IC pin, it is attached to the pin. When I connect a wire _to_ an IC pin, it is connected to perfoboard eye instead. As a single wire between two ICs always has a _to_ side, this gives the arbitrary decision which IC the wire should connect to (aka the IC you more likely need to move at some point of design) and which side you connect to board, having the IC only connected as long as it stays in place.

vanepp, 2023-04-11:
As you see above what you want happens if the wire terminates on the IC pin rather than on the perf board (although that doesn't reflect what typically happens in real life!) Doing that always would likely be difficult (because the perf board and the part are different connections) and more importantly would preclude what is currently possible which is moving an IC that was incorrectly placed although that can have unexpected effects and thus isn't recommended. Here is an example of what can happen. Starting from here where the IC is connected to the perf board pads. 

[image]

I can correct a placement error by moving the IC one space left 

[image]

which I can't do without moving all the wires on the IC probably by deleting and replacing them to make sure they connect to the IC and not the perf board. In the case where the wires are connected to the pins, moving the IC can then create unexpected (and wrong!) connections as the IC will attach to the perf board. Here I connected the wires to the pins

[image]

then moved the IC (with the wires following.) note when I click on a pad on the resistor perfboard strip all connections (including a new one in the ICs new position) light up as being connected together. 

[image]

If there were already components attached to that perfboard strip you would potentially create a short across a component like this

[image]

Here we see the resistor has shorted because of the added wire between two perfboard traces and the movement and re connection of the IC to the perf board.  That can still happen now, but it requires that you make the connection to the IC pin (which is uncommon) rather than to the perfboard which is the usual case. As a result I expect there will be reluctance to make such a change (although that isn't my decision.)

dronus, 2023-05-11:
I think what you are showing is a "strip board" in my nomenclature. What I am talking about is a board with solder dots only, no connections at all, those will be made by wires. So there have wires always connect to pins would be a great benefit, as it would allow move any part around without changing the wiring between them.

dronus, 2023-05-11:
Even if this is meant for better experience on strip board, it's still abitrary why the "from" connection point is attached to the part, but the "to" connection point is attached to the board.

dronus, 2023-05-20:
I think it depends on the order of the clicks. The first click seems to always attach to a _pin_, the second one to a donut.
So there is _some_ control, but for any single wire, only _one_ end can be connected to a pin, the other will always connect to the board. That gives headaches on the order of clicks, because if well thought of, some components may be moved later without breaking the netlist, but others not.

KjellMorgenstern, 2024-08-01:
We had a closer look at this. Proximity or mouse drop event timings don't play a role here.

If a place has multiple possible connections, they are sorted according to the the method "wireLessThan" in connectoritem.cpp.

Bascially that is:
- Female connectors first (the perfboard or breadboard)
- Then male connectors.
- Then pads.
- Then all others.

For connectors of same type, the z-Order is used.

KjellMorgenstern, 2024-08-01:
Considering this as a bug. 1. We want to have the control over the target of the connection, same as we have it when starting a wire. 

Besides this, because the highlight elements, the hit box etc all increase in size when zooming in, it still is tricky to select the element, even at high zoom levels. This is a different issue, we already have it at high priority.

closed 2024-10-10 by KjellMorgenstern