fritzing/fritzing-app #4013: custom part with no layerId in pcb does not get exported to the sketch.

vanepp, 2023-03-16:
## Current Behaviour

I just ran across a particularly nasty (if fortunately rare!) bug in a part. If you leave off the layerId in the pcb layer (which is breadboard in this case to suppress pcb) the part does not export when saved in a sketch, but no error is presented when the sketch is saved. Create a sketch with a DM860A Stepper Motor Driver (this one the current version is fixed now!) 

[DM860A Stepper Motor Driver-broken.fzpz.zip](https://github.com/fritzing/fritzing-app/files/10996550/DM860A.Stepper.Motor.Driver-broken.fzpz.zip) 

remove the trailing .zip to get the .fzpz file. Which has a pcb definition like this which is missing its layerId

```
    <pcbView>
      <layers image="breadboard/prefix0000_911c140c21fa07989d9ddccd6943645a_1_breadboard.svg">
   </layers>
```
instead of the correct

```
    <pcbView>
      <layers image="breadboard/prefix0000_911c140c21fa07989d9ddccd6943645a_1_breadboard.svg">
        <layer layerId="breadboard"/>
      </layers>
    </pcbView>
```

The result looks like this (all looks fine in Fritzing!)

[image]
 
saving the sketch has no errors (but doesn't work correctly as the custom part is not exported in the .fz file of the sketch!) 

[image]

however when you reload the sketch in to Fritzing there is a problem:

[image]

Fritzing is complaining about not finding the custom part's fzp file in the .fz file for the sketch (presumably because it wasn't loaded.) Hitting OK completes the load but omits the custom part. 

[image]

**Build:**
Version 0.9.10 
(bCD-2134-0-40d23c29 2022-07-01) 64 [Qt 5.15.2]

**Operating System:**

Windows 10

**Steps to reproduce:**

see Current Behaviour

## Expected Behaviour

I guess the best bet would be an error message on saving the sketch that a custom part (preferably named) was not able to be saved and thus the sketch will not load correctly until the part is fixed. I don't see anything else that can be done at this point, but at least the user will be aware that there is something wrong before they try and reload the sketch. Even better if it is possible would be to ignore the lack of a layerId and save the part in the .fz file anyway, but there may be problems with that.  The part has been this way since 2017 and this is the first time (Mar of 2023) I have seen a complaint in the forums about this problem. I've replaced the part with a fixed one, but there are lots of folks using this part out there it appears.

KjellMorgenstern, 2023-03-17:
Thank you a lot, good catch. This might explain multiple of the "Unable to find the following part" reports we have seen.
It should be simple to add a check to github/fritzing/fritzing-parts CI pipeline (if it isn't already there, not sure out of my head)

I tend to fix it by immediately rejecting such parts, so Fritzing will not load them at all.
Allowing this with some "ah, probably you meant this layerid which we discovered in the svg") will make Fritzing even harder to maintain.

vanepp, 2023-03-17:
"I tend to fix it by immediately rejecting such parts, so Fritzing will not load them at all.
Allowing this with some "ah, probably you meant this layerid which we discovered in the svg") will make Fritzing even harder to maintain."

That sounds like the best solution, I agree that rejecting the part on load will fix all the problems, the user will be aware that the part is broken. It will still leave the current broken parts in sketches but is probably the better solution. I'm just about to add some more issues to the routing database corruption issue from this same incident. He has routing database corruption which is doing things such as moving locked parts as well as the part issue.