fritzing/fritzing-app #4015: Unable to build on macOS Monterey

Judder, 2023-03-21:
## Current Behaviour

alexander@Alexs-iMac fritzing-app % ./tools/deploy_fritzing_mac.sh                
>> working directory
/Users/alexander/Desktop/fritzing/fritzing-app
>> deploy directory
/Users/alexander/Desktop/fritzing/fritzing-app/../deploy-app
>> build directory
/Users/alexander/Desktop/fritzing/fritzing-app/../release64
>> building fritzing from working directory

<build commands>

clang: error: no such file or directory: '/usr/lib/libz.dylib'
clang: error: no such file or directory: '/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation'
clang: error: no such file or directory: '/System/Library/Frameworks/Carbon.framework/Carbon'
clang: error: no such file or directory: '/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit'
clang: error: no such file or directory: '/System/Library/Frameworks/Security.framework/Versions/A/Security'
make[1]: *** [../release64/Fritzing.app/Contents/MacOS/Fritzing] Error 1
make: *** [release] Error 2

**Build:**
v0.9.9

commit f0af53a9077f7cdecef31d231b85d8307de415d4

**Operating System:**
macOS Monterey

**Steps to reproduce:**
- Checkout from git
- Install as per https://github.com/fritzing/fritzing-app/wiki/1.1-Mac-notes with Qt Creator
- Compile

**Reference posts of similar behaviour**
https://forum.fritzing.org/t/diary-of-a-mac-m1-build/12657/2
https://forum.fritzing.org/t/building-the-latest-fritzing-on-macos-monterey/16760

## Expected Behaviour
Successful build

jklap, 2023-03-23:
You can try this:

```
diff --git a/phoenix.pro b/phoenix.pro
index dab1ba21..83876804 100644
--- a/phoenix.pro
+++ b/phoenix.pro
@@ -90,10 +90,10 @@ macx {
     QMAKE_INFO_PLIST = FritzingInfo.plist
     #DEFINES += QT_NO_DEBUG                # uncomment this for xcode
     LIBS += -lz
-    LIBS += /usr/lib/libz.dylib
-    LIBS += /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
-    LIBS += /System/Library/Frameworks/Carbon.framework/Carbon
-    LIBS += /System/Library/Frameworks/IOKit.framework/Versions/A/IOKit
+#    LIBS += /usr/lib/libz.dylib
+    LIBS += -framework corefoundation
+    LIBS += -framework carbon
+    LIBS += -framework iokit
     LIBS += -liconv
 }
 unix {
diff --git a/pri/libgit2detect.pri b/pri/libgit2detect.pri
index 66b3e0d0..b425dd1c 100644
--- a/pri/libgit2detect.pri
+++ b/pri/libgit2detect.pri
@@ -59,7 +59,7 @@ unix {
             error("static libgit2 library not found in $$LIBGIT2LIB")
         }
         macx {
-            LIBS += $$LIBGIT2LIB/libgit2.a /System/Library/Frameworks/Security.framework/Versions/A/Security
+            LIBS += $$LIBGIT2LIB/libgit2.a -framework security
         } else {
             LIBS += $$LIBGIT2LIB/libgit2.a  -lssl -lcrypto
         }
```

Caveat that I haven't tested it other than a compile and then seeing if the app opens

Judder, 2023-04-12:
What release of QT are you using for your build?

Fritzing seems to only be compatible with 5.x but I'm now having trouble linking in quazip as even make it for QT 5.x it lacks the /lib and /include structure that pri/quazipdetect.pri is looking for

`cmake -S . -B /usr/local/Cellar/quazip/latest -DCMAKE_PREFIX_PATH=/Users/alexander/Qt/5.15.2/clang_64 -D QUAZIP_QT_MAJOR_VERSION=5`

Trying to use `quazip1_qt6` gives errors due to incompatability

`:-1: error: Undefined symbols for architecture x86_64:
  "QuaZipFile::open(QFlags<QIODevice::OpenModeFlag>)", referenced from:
      FolderUtils::unzipTo(QString const&, QString const&, QString&) in folderutils.o
  "QuaZipFile::open(QFlags<QIODevice::OpenModeFlag>, QuaZipNewInfo const&, char const*, unsigned int, int, int, bool, int, int, int)", referenced from:
      FolderUtils::createZipAndSaveTo(QDir const&, QString const&, QStringList const&) in folderutils.o`

mr-canoe-head, 2023-04-24:
> You can try this:

I can confirm this worked for me (although I  have not yet thoroughly tested the app). Thanks for providing the fix.

closed 2023-11-03 by KjellMorgenstern