fritzing/fritzing-app #4018: Allow fzpz in fritzing-parts

KjellMorgenstern, 2023-03-24:
## Problem
Parts a currently structured in folders for fzp and layers.
This structure allows to reuse certain standardized footprints. However, custom parts that come with their own footprint and symbols are a very common use case. For these cases, the organization into folders and sharing of files causes complexity that is hard to manage and not necessary.

## Proposed Solution
Fritzing app should allow fzpz in fritzing-parts/fzpz in arbitrary subfolders

failiz, 2023-03-24:
I agree with simplifying the organization requirements for parts that have all custom SVG files. However, I do not think we should store fzpz files in the repository as they are compressed and version control would be useless. Maybe we could create a folder for each file where the fpz and SVGs files are stored in plain text.

vanepp, 2023-03-24:
I tend to agree with @failiz  here, I would prefer to keep the repository to uncompressed files. I think that this may have been what the contrib directories were intended for (or could be successfully repurposed for anyway.)  Here is the current repository layout, the contrib subtree is present but not apparently used

[image]

[image]

[image]

a loader modification that would unpack the user's .fzpz file in to this tree may be the best answer here I think, as it leaves the repository uncompressed but perhaps does what Kjell is looking for? I'm not sure what implications parts in those directories have, I assume they are searched when looking for parts, but that may not have been implemented yet when development stopped.

KjellMorgenstern, 2025-07-01:
Reviewing this again: fzpz might be a bit cleaner than spreading part files over core and svg directories, but I think this issue is can also be solved with proper tools, like the checks in the CI pipeline.

 -> Low Priority or even Close this issue ?

closed 2025-07-28 by KjellMorgenstern