fritzing/fritzing-app #4020: Bendable leg parts won't extend the leg when locked.

vanepp, 2023-04-02:
## Current Behaviour

Drag a resistor in to breadboard view

[image]

click on the connector pin and drag it, leg extends as expected

[image]

delete that and drag in a new resistor then lock the part

[image]

click on the connector and attempt to drag it and it doesn't move but remains at its current position, so locking the part is causing the leg to not move (which I think is a bug!)

[image]

**Build:**

Version 0.9.10 
(bCD-2134-0-40d23c29 2022-07-01) 64 [Qt 5.15.2]

**Operating System:**
Windows 10

**Steps to reproduce:**

see current behaviour

## Expected Behaviour

The leg should extend as normal, only moving the body of the part should be blocked by the lock, not extending the leg.

KjellMorgenstern, 2023-04-02:
I think it is expected that the legs are locked together with the part. I didn't look it up, but there must be some code for bendable legs that checks the lock state. 

Is there any way this behavior is inconsistent? Or, why would it be more useful to still have the legs bendable, and not be able to lock them?

vanepp, 2023-04-03:
I think that the legs should still move, I think the intent is (or should be anyway) that the part itself not move but the legs still can. The idea of lock is that the part won't move when you are wiring (and accidentally select the part) but I don't think it should apply to the bendable legs, just the location if the part body (which may be a difficult problem given the odd configuration of bendable leg parts though!) It likely isn't deadly if it isn't easy to fix, but I at least think it is counterintuitive.