fritzing/fritzing-app #4021: Locked part symbol on headers obsucured

vanepp, 2023-04-04:
## Problem

[image]

The locked symbol on this single pin header is lost in the dark of the part body. It is hard to see if a header is locked without right clicking on it and looking at the lock status (which is a pain!) 

## Proposed Solution

Move the locked image in the direction of the green arrow so it is at least partly outside the connector body and thus more visible like the resistor next to it which is also locked.

failiz, 2023-06-04:
Related to #1999
Maybe the lock symbol position could optionally be defined in the part file. I guess that finding a good place for all the parts by an algorithm is going to be hard.
Another solution would be to place the symbol on the top layer, it seems that the legs of the resistor are on top of the lock symbol.
Adding a background to the lock symbol could also help to deal with the low contrast on black parts.

KjellMorgenstern, 2023-06-04:
Maybe the lock symbol could heighlight or even zoom in for a moment when trying to modify the part.
I am not sure if this works in all cases. But right now, you could even miss the symbol if properly placed, and keep search on how to get the desired modification, especially when not yet familar with Fritzing.

vanepp, 2023-06-05:
> @failiz  Maybe the lock symbol position could optionally be defined in the part file. I guess that finding a good place for all the parts by an algorithm is going to be hard.
Another solution would be to place the symbol on the top layer, it seems that the legs of the resistor are on top of the lock symbol.
Adding a background to the lock symbol could also help to deal with the low contrast on black parts.

I'd rather avoid a change to each part if possible (but admit to an incomplete understanding of the layers and how they interact.) The top layer and light background on the lock icon to deal with dark part backgrounds both sound easy and very useful to me. I'd be in favor of trying that first to see if it does enough before adding a modification to every part. I'm a fan of easy (lazy to the core!) @KjellMorgenstern Again if it is easy, a highlight of the lock symbol when you try and modify a part would be useful. Anything to draw the eye when something doesn't work as expected should be useful! That should trigger though and hopefully research in to what the lock symbol means (in some cases nothing is going to trigger thought though) as an indication that the lock symbol is the problem.