fritzing/fritzing-app #4023: Redundant entry in fz file.

vanepp, 2023-04-12:
## Current Behaviour

[image]

From this test sketch (part of poking at the database corruption issue) we appear to have a redundant entry for the wire.  This is new sketch with a Raspberry PI2 with a single wire connecting to the breadboard (for use as a reference to a corrupted .fz file, which I have managed to reduce to only these two connections.) When saved as a .fzz file here

[Pi-Motion-Sensor-5.fzz.zip](https://github.com/fritzing/fritzing-app/files/11216391/Pi-Motion-Sensor-5.fzz.zip)

(remove the trailing .zip to keep github happy to get the .fzz file)

and then uncompressing the .fzz file and editing Pi-Motion-Sensor-5.fz we see there is an extra (and I expect unneeded, although that may be incorrect!) instance of the wire. 

``        <instance moduleIdRef="WireModuleID" modelIndex="7011" path=":/resources/parts/core/wire.fzp">
            <views/>
        </instance>``

which is followed immediately by the complete wire definition (with the same moduleRef and ModelIndex) which I expect may cause indexing issues although it doesn't seem to!

``
        <instance moduleIdRef="WireModuleID" modelIndex="7011" path=":/resources/parts/core/wire.fzp">
            <title>Wire1</title>
            <views>
                <breadboardView layer="breadboardWire">
                    <geometry z="3.5" x="243.001" y="-0.00117619" x1="0" y1="0" x2="-197.999" y2="0.00115863" wireFlags="64"/>
                    <wireExtras mils="22.2222" color="#418dd9" opacity="1" banded="0"/>
                    <connectors>
                        <connector connectorId="connector1" layer="breadboardWire">
                            <geometry x="0" y="0"/>
                            <connects>
                                <connect connectorId="connector5" modelIndex="7008" layer="breadboardbreadboard"/>
                            </connects>
                        </connector>
                        <connector connectorId="connector0" layer="breadboardWire">
                            <geometry x="0" y="0"/>
                            <connects>
                                <connect connectorId="pin43J" modelIndex="5785" layer="breadboardbreadboard"/>
                            </connects>
                        </connector>
                    </connectors>
                </breadboardView>
                <schematicView layer="schematicTrace">
                    <geometry z="5.5" x="237.838" y="-4.364" x1="0" y1="0" x2="-201.471" y2="3.63667" wireFlags="64"/>
                    <wireExtras mils="9.7222" color="#404040" opacity="1" banded="0"/>
                    <connectors>
                        <connector connectorId="connector1" layer="schematicTrace">
...`` 

as noted this doesn't seem to hurt anything and may be required for some reason that I don't understand, but if not eliminating the duplicate may be a good bet both to reduce clutter and file size inf .fz files with a lot of wires. 

**Build:**

Version 0.9.10 
(bCD-2134-0-40d23c29 2022-07-01) 64 [Qt 5.15.2]

**Operating System:**
Windows 10

**Steps to reproduce:**

see current behaviour

## Expected Behaviour

omit the first incomplete version of the entry to reduce clutter and fxz file size (low priority though I expect!)

KjellMorgenstern, 2023-05-10:
This indeed looks like #3904  . When the issue happens, not always are ratsnest lines shown. The connection does not get properly cleaned from the breadboard. Wether a ratsnest line shows depends on if there is another part connected to the same breadboard row.

vanepp, 2023-05-19:
It still occurs on Fritzing 1.0. As part of my script testing I created this .fzz file in Fritzing1.0 expecting a clean output

[testcase-1wire-fritzing1_0.fzz.zip](https://github.com/fritzing/fritzing-app/files/11513463/testcase-1wire-fritzing1_0.fzz.zip)

(as usual delete the trailing .zip to get the .fzz file, then unzip the .fzz to get the .fz file) 

which looks like this

[image]

To my surprise I get this:

$ fzcheck.py -c testcase-1wire-fritzing1_0.fz

***    Process testcase-1wire-fritzing1_0.fz    ***

5783 TwoLayerRectanglePCBModuleID (srcln 13) no connections
5793 WireModuleID (srcln 124) no connections
error: modelindex 5793 already exists at srcline 124
                       found again at srcline 127, overwritten

*** connections

5787 generic_female_pin_header_2_100mil (srcln 24)
  5787-connector1-breadboard 5793-connector1-breadboardWire (srcln 31,34)
  5787-connector1-schematic 5793-connector1-schematicTrace (srcln 48,51)
  5787-connector1-copper0 5793-connector1-copper1trace (srcln 64,67)

5793 WireModuleID (srcln 127)
  5793-connector1-breadboardWire 5787-connector1-breadboard (srcln 134,137)
  5793-connector1-schematicTrace 5787-connector1-schematic (srcln 152,155)
  5793-connector1-copper1trace 5787-connector1-copper0 (srcln 170,173)
  5793-connector0-breadboardWire 5790-connector0-breadboard (srcln 140,143)
  5793-connector0-schematicTrace 5790-connector0-schematic (srcln 158,161)
  5793-connector0-copper1trace 5790-connector0-copper0 (srcln 176,179)
  5793-connector1-breadboardWire 5787-connector1-breadboard (srcln 134,137)
  5793-connector0-breadboardWire 5790-connector0-breadboard (srcln 140,143)
  5793-connector1-schematicTrace 5787-connector1-schematic (srcln 152,155)
  5793-connector0-schematicTrace 5790-connector0-schematic (srcln 158,161)
  5793-connector1-copper1trace 5787-connector1-copper0 (srcln 170,173)
  5793-connector0-copper1trace 5790-connector0-copper0 (srcln 176,179)

5790 generic_female_pin_header_2_100mil (srcln 74)
  5790-connector0-breadboard 5793-connector0-breadboardWire (srcln 81,84)
  5790-connector0-schematic 5793-connector0-schematicTrace (srcln 98,101)
  5790-connector0-copper0 5793-connector0-copper1trace (srcln 114,117)

*** Would be deleted

  5793 WireModuleID (srcln 124)

which is because the error reported in this issue is still there in the .fz file

```
<?xml version="1.0" encoding="UTF-8"?>
<module fritzingVersion="1.0.0rc.2023-05-11.CD-4028-0-eb6a9551">
... (this entry shouldn't be here I don't think!) 
        <instance moduleIdRef="WireModuleID" modelIndex="5793" path=":/resources/parts/core/wire.fzp">
            <views/>
        </instance>
(because it is identical to this correct and complete entry which immediately follows it and thus is redundant, which is why the script is suggesting removing it!) 
        <instance moduleIdRef="WireModuleID" modelIndex="5793" path=":/resources/parts/core/wire.fzp">
            <title>Wire1</title>
            <views>
                <breadboardView layer="breadboardWire">
                    <geometry z="3.5" x="315" y="27" x1="0" y1="0" x2="-117" y2="0" wireFlags="64"/>
                    <wireExtras mils="22.2222" color="#418dd9" opacity="1" banded="0"/>
                    <connectors>
                        <connector connectorId="connector1" layer="breadboardWire">
                            <geometry x="0" y="0"/>
                            <connects>
                                <connect connectorId="connector1" modelIndex="5787" layer="breadboard"/>
                            </connects>
                        </connector>
                        <connector connectorId="connector0" layer="breadboardWire">
                            <geometry x="0" y="0"/>
                            <connects>
                                <connect connectorId="connector0" modelIndex="5790" layer="breadboard"/>
                            </connects>
                        </connector>
                    </connectors>
                </breadboardView>                                                   
```

KjellMorgenstern, 2023-05-25:
Thanks, this clear way to reproduce this issue helped a lot. 
I think we found a solution. There is a log message about DragWires deleted, but the model is not deleted along with that part. When deleting the according model, the redundant entries don't appear.

Not  a fix we can add to 1.0.0 so short before the release, though.

vanepp, 2023-09-21:
As of 1.0.1 I have no longer seen this error when running my script (lots of others, but not this one!) so I will close this issue and start a new issue if I start seeing it again (which seems unlikely.)

closed 2023-09-21 by vanepp