fritzing/fritzing-app #4025: Simply doesn't work: Components selection leads to a disabled mouse icon.

bhabegger, 2023-05-06:
I just downloaded Fritzing after paying the 8€ fee to be able to use the simulator... But this versions doesn't work where the opensource version does... See the attached video.

[Screencast from 2023-05-06 18-41-14.webm](https://user-images.githubusercontent.com/1582356/236636750-125e8b72-c65e-4cdd-8fcf-0b7514ff119d.webm)

KjellMorgenstern, 2023-05-06:
Could you add the linux version, window manager version and 3rd party fritzing build version?

KjellMorgenstern, 2023-05-07:
Seems like this is not a Fritzing bug, but the linux distribution, and affects a plethora of applications.

https://bugs.launchpad.net/ubuntu/+source/mutter/+bug/2013235

https://gitlab.gnome.org/GNOME/mutter/-/issues/2715

https://forum.qt.io/topic/144831/recent-upgrade-to-ubuntu-23-04-breaks-qt-designer?lang=de

dvarrel, 2023-05-17:
yes , same issue with ubuntu 23.04 ( works fine before with ubuntu 22.04 )

KjellMorgenstern, 2023-05-17:
We experimented, but it seems the solution would then break Fritzing for all other users (eg. ubuntu 22.04). 
One workaround if you don't want to downgrade is using Xfce.

dvarrel, 2023-05-17:
how do you break Fritzing for all other users ?

KjellMorgenstern, 2023-05-17:
adding native wayland support

dvarrel, 2023-05-17:
seems 23.04 use wayland no ?

dvarrel, 2023-05-17:
fritzing is a appimage,, so i can't choose the qt library wayland

KjellMorgenstern, 2023-05-17:
I didn't suggest wayland, but Xfce.

bhabegger, 2023-05-28:
And how is using xfce done ?

KjellMorgenstern, 2023-05-28:
Install xfce with your packet manager. Then choose Xfce in the login screen:

[image]

[image]

KjellMorgenstern, 2023-05-28:
Another workaround is described here, compiling libmutter from source. It was confirmed to work on all Qt applications, so should also work for Fritzing.
Note that you have to repeat the steps for each update to libmutter until the fix is included by ubuntu.

https://github.com/OpenShot/openshot-qt/issues/5221#issuecomment-1546539610

dvarrel, 2023-05-30:
It works, thanks you very much !!!

KjellMorgenstern, 2023-07-18:
The issue was fixed in mutter 44.2
https://bugs.launchpad.net/ubuntu/+source/mutter/+bug/2013235

Which was release with "lunar-updates" for 23.04:
https://packages.ubuntu.com/search?keywords=mutter&searchon=names&suite=lunar-updates&section=all