fritzing/fritzing-app #4026: Silk Screened Logo does not export

Moonbase1, 2023-05-11:
## Current Behavior
Text
<please attach screenshot if possible>
**Build:**
0.9.10 
**Operating System:**
Windows 11

**Steps to reproduce:**
- Load SVG as silkscreen 
- it works in fritzing 
- it does not export 

## Expected Behavior
I had two SVG files, one is a logo and the other is a QR code for PCB instructions. The QR code works correctly, following the identical procedure to bring in the logo is not exporting. Fritzing is showing the logo, when I export as a png, gerber file, or even using the built in function to export it to Aisling for production only the QR code works. This has happened with 2 other logo files as well, tried PNG and SVG. Only the QR code is functional.

KjellMorgenstern, 2023-05-11:
Please don't forget to attach sample SVGs. Also, SVG was extended for web, it can contain CSS, animations, references to external files...

Fritzing only supports TinySVG, not the full blown SVG 2.0 . Make sure to select the right format when exporting the SVG from you editor.

Moonbase1, 2023-05-11:
> 

It's the same exact file type as the QR code. The QR code exports correctly. It does this with both PNG and SVG.

KjellMorgenstern, 2023-05-11:
There is not enough information to act on this, so I have to close this for now. Feel free to reopen if you have additional information that could help to test the issue.

closed 2023-05-11 by KjellMorgenstern