fritzing/fritzing-app #4027: Uninstall Fritzing doesn't remove all files.

vanepp, 2023-05-12:
## Current Behaviour

Uninstall Fritzing 0.9.10 (all users) to test a single user install. Remove appeared to work but left a single svg file in the parts directory

owner@owner-PC /cygdrive/c/program files/fritzing
$ ls -lR
.:
total 4
drwxrwx---+ 1 SYSTEM SYSTEM 0 May 11 18:07 fritzing-parts

./fritzing-parts:
total 0
drwxrwx---+ 1 SYSTEM SYSTEM 0 May 11 18:07 svg

./fritzing-parts/svg:
total 4
drwxrwx---+ 1 SYSTEM SYSTEM 0 May 11 18:07 core

./fritzing-parts/svg/core:
total 640
drwxrwx---+ 1 SYSTEM SYSTEM 0 May 11 18:07 schematic

./fritzing-parts/svg/core/schematic:
total 4
-rwxr-x---+ 1 SYSTEM SYSTEM 2123 Jul  5  2022 dcpower1.svg

This likely isn't deadly, but it also isn't correct (but I expect low priority)

From this forum post

https://forum.fritzing.org/t/fritzing-runs-after-install-but-not-after-closing-and-reopening/19209

**Build:**
Version 0.9.10 
(bCD-2134-0-40d23c29 2022-07-01) 64 [Qt 5.15.2]

**Operating System:**
Windows 10

**Steps to reproduce:**
- Uninstall Fritzing 0.9.10
- Check the program files\fritzing directory, fritzing parts is still present. 

## Expected Behaviour

Deleting the entire parts directory (in fact likely the entire fritzing directory.)

KjellMorgenstern, 2023-05-16:
@vanepp Did you check if the file existed before the 0.9.10 was installed? If the file is a remain of and old or unknown installation, the installer might play save and not remove it.

vanepp, 2023-05-16:
> Did you check if the file existed before the 0.9.10 was installed? If the file is a remain of and old or unknown installation, the installer might play save and not remove it.

No I didn't think to check that, and it looks to be the correct answer I just unintalled fritzing-1.0.0-l4028-eb6a9551-win64 (which uninstalled completely) then reinstalled and uninstalled 0.9.10 which also now uninstalls completely so I expect that is correct the file existed and was preserved. I'll close this.

closed 2023-05-16 by vanepp